groeck/nct6775 #28: compiling the driver for centos 5.10

limort, 2015-01-01:
I'm trying to compile the driver for kernel 2.6.18-371.el5PAE and I'm facing some compilation problems:
make[1]: Warning: File `/usr/src/kernels/2.6.18-371.el5-PAE-i686/arch/i386/Makefile.cpu' has modification time 4e+08 s in the future
make[2]: Warning: File `scripts/Makefile.lib' has modification time 1.9e+08 s in the future
  CC [M]  /root/nct6775-master/nct6775.o
/root/nct6775-master/nct6775.c: In function ‘nct6775_create_attr_group’:
/root/nct6775-master/nct6775.c:993: error: ‘struct attribute_group’ has no member named ‘is_visible’
/root/nct6775-master/nct6775.c: At top level:
/root/nct6775-master/nct6775.c:3147: error: unknown field ‘is_visible’ specified in initializer
/root/nct6775-master/nct6775.c:3147: warning: excess elements in struct initializer
/root/nct6775-master/nct6775.c:3147: warning: (near initialization for ‘nct6775_group_other’)
/root/nct6775-master/nct6775.c: In function ‘nct6775_probe’:
/root/nct6775-master/nct6775.c:3315: error: implicit declaration of function ‘dev_get_platdata’
/root/nct6775-master/nct6775.c:3315: warning: initialization makes pointer from integer without a cast
/root/nct6775-master/nct6775.c:3998: warning: assignment from incompatible pointer type
/root/nct6775-master/nct6775.c: In function ‘nct6775_remove’:
/root/nct6775-master/nct6775.c:4016: warning: passing argument 1 of ‘hwmon_device_unregister’ from incompatible pointer type
/root/nct6775-master/nct6775.c: At top level:
/root/nct6775-master/nct6775.c:4111: error: variable ‘nct6775_dev_pm_ops’ has initializer but incomplete type
/root/nct6775-master/nct6775.c:4112: error: unknown field ‘suspend’ specified in initializer
/root/nct6775-master/nct6775.c:4112: warning: excess elements in struct initializer
/root/nct6775-master/nct6775.c:4112: warning: (near initialization for ‘nct6775_dev_pm_ops’)
/root/nct6775-master/nct6775.c:4113: error: unknown field ‘resume’ specified in initializer
/root/nct6775-master/nct6775.c:4113: warning: excess elements in struct initializer
/root/nct6775-master/nct6775.c:4113: warning: (near initialization for ‘nct6775_dev_pm_ops’)
/root/nct6775-master/nct6775.c:4114: error: unknown field ‘freeze’ specified in initializer
/root/nct6775-master/nct6775.c:4114: warning: excess elements in struct initializer
/root/nct6775-master/nct6775.c:4114: warning: (near initialization for ‘nct6775_dev_pm_ops’)
/root/nct6775-master/nct6775.c:4115: error: unknown field ‘restore’ specified in initializer
/root/nct6775-master/nct6775.c:4115: warning: excess elements in struct initializer
/root/nct6775-master/nct6775.c:4115: warning: (near initialization for ‘nct6775_dev_pm_ops’)
/root/nct6775-master/nct6775.c:4127: error: unknown field ‘pm’ specified in initializer
/root/nct6775-master/nct6775.c:4127: warning: initialization from incompatible pointer type
/root/nct6775-master/nct6775.c:4135: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘__initconst’
/root/nct6775-master/nct6775.c: In function ‘nct6775_find’:
/root/nct6775-master/nct6775.c:4208: error: ‘nct6775_sio_names’ undeclared (first use in this function)
/root/nct6775-master/nct6775.c:4208: error: (Each undeclared identifier is reported only once
/root/nct6775-master/nct6775.c:4208: error: for each function it appears in.)
/root/nct6775-master/nct6775.c: In function ‘sensors_nct6775_init’:
/root/nct6775-master/nct6775.c:4267: error: implicit declaration of function ‘acpi_check_resource_conflict’
make[2]: *** [/root/nct6775-master/nct6775.o] Error 1
make[1]: *** [_module_/root/nct6775-master] Error 2
make: *** [modules] Error

groeck, 2015-01-01:
On Thu, Jan 01, 2015 at 02:31:52AM -0800, limort wrote:
> I'm trying to compile the driver for kernel 2.6.18-371.el5PAE and I'm facing some compilation problems:
> make[1]: Warning: File `/usr/src/kernels/2.6.18-371.el5-PAE-i686/arch/i386/Makefile.cpu' has [...]

Sorry, that kernel version is just too old. 

Guenter

groeck, 2015-01-17:
I updated compat.h to reflect the oldest still supported kernel (2.6.32).

closed 2015-01-17 by groeck

Dmitry1987, 2015-02-03:
Hi,
I am too trying to make this module work, and want to ask if it's possible to build the module on newer kernel (or using 2.6.32 kernel-headers) and use it still with 2.6.18?
or it's better not to waste the time and this particular driver will not work on 2.6.18 in any case?

Thanks!

graysky2, 2015-02-03:
Kernel headers and matched to kernel source; you can't build against a non-matched set and get a binary that will work with either one.

groeck, 2015-02-03:
On 02/03/2015 02:05 AM, Dmitry wrote:
> Hi,
> I am too trying to make this module work, and want to ask if it's possible to build the module on newer kernel (or using 2.6.32 kernel-headers) and use it still with 2.6.18?
> or it's better not to waste the time and this particular driver will not work on 2.6.18 in any case?
>

You could try with older versions of the driver, but I have no idea if they work.

Guenter