groeck/nct6775 #29: CentOS 7 Compile errors

CoMMyz, 2015-05-07:
Hi,

I am attempting to compile on CentOS 7 with kernel 3.10.0-229.1.2.el7.x86_64 and i get the following compile issue. Can you assist?

Thanks

  CC [M]  /data/os/root/nct6775/nct6775.o
In file included from /data/os/root/nct6775/nct6775.c:63:0:
/data/os/root/nct6775/compat.h:23:12: error: static declaration of ‘sysfs_create_groups’ follows non-static declaration
 static int sysfs_create_groups(struct kobject *kobj,
            ^
In file included from include/linux/kobject.h:21:0,
                 from include/linux/module.h:16,
                 from /data/os/root/nct6775/nct6775.c:49:
include/linux/sysfs.h:223:18: note: previous declaration of ‘sysfs_create_groups’ was here
 int __must_check sysfs_create_groups(struct kobject *kobj,
                  ^
In file included from /data/os/root/nct6775/nct6775.c:63:0:
/data/os/root/nct6775/compat.h:43:13: error: static declaration of ‘sysfs_remove_groups’ follows non-static declaration
 static void sysfs_remove_groups(struct kobject *kobj,
             ^
In file included from include/linux/kobject.h:21:0,
                 from include/linux/module.h:16,
                 from /data/os/root/nct6775/nct6775.c:49:
include/linux/sysfs.h:229:6: note: previous declaration of ‘sysfs_remove_groups’ was here
 void sysfs_remove_groups(struct kobject *kobj,
      ^
In file included from /data/os/root/nct6775/nct6775.c:63:0:
/data/os/root/nct6775/compat.h:54:32: error: redefinition of ‘PTR_ERR_OR_ZERO’
 static inline int __must_check PTR_ERR_OR_ZERO(__force const void *ptr)
                                ^
In file included from /usr/src/kernels/3.10.0-229.1.2.el7.x86_64/arch/x86/include/asm/processor.h:30:0,
                 from /usr/src/kernels/3.10.0-229.1.2.el7.x86_64/arch/x86/include/asm/thread_info.h:22,
                 from include/linux/thread_info.h:54,
                 from include/linux/preempt.h:9,
                 from include/linux/spinlock.h:50,
                 from include/linux/seqlock.h:35,
                 from include/linux/time.h:5,
                 from include/linux/stat.h:18,
                 from include/linux/module.h:10,
                 from /data/os/root/nct6775/nct6775.c:49:
include/linux/err.h:55:32: note: previous definition of ‘PTR_ERR_OR_ZERO’ was here
 static inline int __must_check PTR_ERR_OR_ZERO(const void *ptr)
                                ^
make[2]: *** [/data/os/root/nct6775/nct6775.o] Error 1
make[1]: *** [_module_/data/os/root/nct6775] Error 2
make: *** [modules] Error 2

groeck, 2015-05-08:
> Hi,
>
> I am attempting to compile on CentOS 7 with kernel 3.10.0-229.1.2.el7.x86_64 and i get the following compile issue. Can you assist?
Just remove the definitions of sysfs_create_groups(),
sysfs_remove_groups(), and PTR_ERR_OR_ZERO() from compat.h.

Looks like CentOS back-ported those functions from a later kernel.
This makes it all but impossible to determine automatically
if the functions need to be provided locally or not.

Guenter

CoMMyz, 2015-05-08:
Confirmed successful compile when removing the mentioned definitions.

Many thanks

closed 2015-05-08 by CoMMyz