groeck/nct6775 #30: No values for NCT6791D on ASUS X99-E WS

lynix, 2015-05-31:
For my **ASUS X99-E WS** motherboard, lm_sensors finds the **NCT6791D** chip

    Driver `nct6775':
      * ISA bus, address 0x290
        Chip `Nuvoton NCT6791D Super IO Sensors' (confidence: 9)

but loading the `nct6775` kernel module does not bring in any sensors or fan nodes in /sys.

Anything I have to do apart from loading the module? I'm using kernel 4.0.4 (on Gentoo). Thanks!

lynix, 2015-05-31:
Some additional info from dmesg:

    [ 5879.459588] nct6775: Found NCT6791D or compatible chip at 0x2e:0x290
    [ 5879.459595] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM_) (20150204/utaddress-254)
    [ 5879.459602] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

groeck, 2015-05-31:
On 05/31/2015 02:18 AM, Alexander Koch wrote:
> Some additional info from dmesg:
>
> |[ 5879.459588] nct6775: Found NCT6791D or compatible chip at 0x2e:0x290
> [ 5879.459595] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts [...]

See "My sensors have stopped working in kernel 2.6.31" at
http://www.lm-sensors.org/wiki/FAQ/Chapter3.

Guenter

lynix, 2015-05-31:
Thanks for the hint, Günter! I've tried the `asus_atk0110` module. Unfortunately, no sensors or fans available with this module either.

So is your suggestion to go risky and add `acpi_enforce_resources=lax` to kernel cmdline?

groeck, 2015-05-31:
On 05/31/2015 11:36 AM, Alexander Koch wrote:
> Thanks for the hint, Günter! I've tried the |asus_atk0110| module. Unfortunately, no sensors or fans available with this module either.
>
> So is your suggestion to go risky and add |acpi_enforce_resources=lax| to kernel cmdline?
>

Your call, really. It does add some risk, but in this case it seems relatively low.

ASUS explicitly doesn't support Linux on their desktop boards, so my more general
advice is to stay away from ASUS if you want to run Linux ;-).

Guenter

lynix, 2015-06-13:
Tried it, worked very well. Only some sensor names missing and a single invalid source:

    nct6775 nct6775.656: Invalid temperature source 31 at index 0, source register 0x100, temp register 0x73

Think I'll live without the readings to be on the safe side for my workstation, and remember your advice regarding ASUS next time ;)  Thanks!

closed 2015-06-13 by lynix

hapylestat, 2015-09-12:
well, possibly Asus is not too friendly to the Linux, but what the hell...this board is the best on x99 chipset ^^

DagNygren, 2016-01-21:
This is also a problem for me running FC23:
Kernel: 4.3.3-300.fc23.x86_64
Motherboard: Intel DP67BG
BIOS version: BGP6710J.86A.2209.2012.0717.2302 (newest available)

setting acpi_enforce_resources=lax| in the boot line made the system unstable and crash about once a week.

groeck, 2016-01-21:
On Thu, Jan 21, 2016 at 07:34:50AM -0800, Dag wrote:
> This is also a problem for me running FC23:
> Kernel: 4.3.3-300.fc23.x86_64
> Motherboard: Intel DP67BG
> BIOS version: BGP6710J.86A.2209.2012.0717.2302 (newest available)
> 
> setting acpi_enforce_resources=lax| in the boot line made the system unstable and crash about once a week.
> 
Sorry, nothing we can do about that. It means that the BIOS 
accesses the chip, and that this access conflicts with the 
access through the kernel driver.

Guenter