groeck/nct6775 #31: CentOS 6.7 Compile errors

Abdula-2015, 2015-08-08:
Hi, dear frend,

I am attempting to compile on CentOS 6.7 with kernel 2.6.32-573.1.1.el6.x86_64 and i get the following compile issue. Can you assist?

Thanks

  CC [M]  /root/sources/nct6775/nct6775-master/nct6775.o
In file included from /root/sources/nct6775/nct6775-master/nct6775.c:63:
/root/sources/nct6775/nct6775-master/compat.h:23: error: redefinition of ‘sysfs_create_groups’
include/linux/sysfs.h:160: note: previous definition of ‘sysfs_create_groups’ was here
/root/sources/nct6775/nct6775-master/compat.h:54: error: redefinition of ‘PTR_ERR_OR_ZERO’
include/linux/err.h:55: note: previous definition of ‘PTR_ERR_OR_ZERO’ was here
/root/sources/nct6775/nct6775-master/nct6775.c: In function ‘nct6775_create_attr_group’:
/root/sources/nct6775/nct6775-master/nct6775.c:994: warning: assignment from incompatible pointer type
/root/sources/nct6775/nct6775-master/nct6775.c: At top level:
/root/sources/nct6775/nct6775-master/nct6775.c:3148: warning: initialization from incompatible pointer type
make[2]: *** [/root/sources/nct6775/nct6775-master/nct6775.o] Error 1
make[1]: *** [_module_/root/sources/nct6775/nct6775-master] Error 2
make: *** [modules] Error 2

pjperry, 2015-08-09:
The following patch should fix these errors on RHEL 6.7

I'll submit a pull request

diff --git a/compat.h b/compat.h
index 5d37fc7..72cea8c 100644
--- a/compat.h
+++ b/compat.h
@@ -20,6 +20,7 @@ u8 vid_which_vrm(void)
 #endif

 #if LINUX_VERSION_CODE < KERNEL_VERSION(3, 12, 0)
+#if !(defined RHEL_MAJOR && RHEL_MAJOR == 6 && RHEL_MINOR >= 7)
 static int sysfs_create_groups(struct kobject *kobj,
                               const struct attribute_group **groups)
 {
@@ -39,6 +40,7 @@ static int sysfs_create_groups(struct kobject *kobj,
        }
        return error;
 }
+#endif

 static void sysfs_remove_groups(struct kobject *kobj,
                                const struct attribute_group **groups)
@@ -51,6 +53,7 @@ static void sysfs_remove_groups(struct kobject *kobj,
                sysfs_remove_group(kobj, groups[i]);
 }

+#if !(defined RHEL_MAJOR && RHEL_MAJOR == 6 && RHEL_MINOR >= 7)
 static inline int __must_check PTR_ERR_OR_ZERO(__force const void *ptr)
 {
        if (IS_ERR(ptr))
@@ -58,7 +61,7 @@ static inline int __must_check PTR_ERR_OR_ZERO(__force const void *ptr)
        else
                return 0;
 }
-
+#endif
 #endif

 #ifdef __NEED_I2C__

pjperry, 2015-08-09:
Pull request created:

https://github.com/groeck/nct6775/pull/32

pjperry, 2015-08-09:
@Abdula-2015  Were you aware that elrepo.org package this driver for EL6 as kmod-nct6775. I've just updated it with the above patch and rebuilt it against EL6.7 so it should be available shortly.

The advantage of using a kmod package is that the driver will work across a range of kernels (as Red Hat try to use a stable kABI) so you won't need to rebuild the driver for each kernel update.

http://elrepo.org/linux/elrepo/el6/x86_64/RPMS/kmod-nct6775-0.0-2.20150809git.el6.elrepo.x86_64.rpm

Thanks to Guenter for maintaining this driver out of tree.

groeck, 2015-08-09:
fixed, thanks to pjperry

closed 2015-08-09 by groeck

Abdula-2015, 2015-08-09:
Thank you very much, my friends!