groeck/nct6775 #33: PWM 1 and 2 controls don't work on MSI Gaming Z97 3

dragonnn, 2015-08-19:
On MSI Z97 Gaming 3 I can not turn fan 1 and 2 off, for fans 3,4,5 it works fine when I do:

echo 1 > pwm5_enable
echo 0 > pwm5

Then fan turns fine off but when I do this on pwm2 the fan goes at full speed and on pwm1 nothing happens. I can provide more feedback like acpi dumps/i2c dumps if this is needed.

groeck, 2015-08-20:
fan 1 and/or 2 may be CPU fans, which can not normally be turned off in the first place. Is this the case ?

dragonnn, 2015-08-20:
Yes, the are CPU fans, can not be this overridden? I have Raijintek Nemesis on my CPU - I am sure it could cool the CPU passive under low load.

dragonnn, 2015-08-20:
http://pastebin.com/TaB2Bitd add log from pwmconfig

groeck, 2015-08-20:
On Thu, Aug 20, 2015 at 07:34:40AM -0700, Mateusz wrote:
> Yes, the are CPU fans, can not be this overridden? I have Raijintek Nemesis on my CPU - I am sure it could cool the CPU passive under low load. 
> 
That is controlled by the fans. I think it is a safety feature.
You might find it in the fine print of the fan manuals somewhere.

You could connect some other fans to those connectors to see if
those can be stopped. But don't leave your CPU fans unconnected.

Guenter

groeck, 2015-08-20:
On Thu, Aug 20, 2015 at 07:55:30AM -0700, Mateusz wrote:
> http://pastebin.com/TaB2Bitd add log from pwmconfig
> 
I have no idea why fan1 is not controllable. The others respond as expected,
with fan2/pwm2 showing the expeted behavior for a cpu fan.

You could move fan1 to another connector and see if the behavior moves
with the fan or with the connector.

Guenter

dragonnn, 2015-08-20:
Hmm I am not sure, but the fan2 have pwm pin and it is set under the pwm-control like other fans the don't have pwm pin and the fan1 is set to dc-control and I can not change it to pwm-control. Well I am sure this is something wrong because fan3-5 don't have pwm control and on pwmX_mode = 1 the works fine.

groeck, 2015-08-20:
On Thu, Aug 20, 2015 at 08:51:55AM -0700, Mateusz wrote:
> Hmm I am not sure, but the fan2 have pwm pin and it is set under the pwm-control like other fans the don't have pwm pin and the fan1 is set to dc-control and I can not change it to pwm-control. Well [...]
You lost me. Your pastebin results seem to suggest that pwm control
on fan3-5 is just fine. What am I missing here ?

Guenter

dragonnn, 2015-08-20:
Yes, the are working fine but I think the pwm_mode is wrong reported by the module - all are 1 expect the pwm1 (with I can not control) and from the documentation pwm_mode 1 means PWM output so fan 2-5 are PWM output and fan1 (with doesn't work) is 0 DC output (0 - 12v) so maybe fan1 should be PWM output too? Problem is I can not change that with echo 1 > pwm1_mode because after echo it is still 0.

groeck, 2015-08-20:
On Thu, Aug 20, 2015 at 10:43:33AM -0700, Mateusz wrote:
> Yes, the are working fine but I think the pwm_mode is wrong reported by the module - all are 1 expect the pwm1 (with I can not control) and from the documentation pwm_mode 1 means PWM output so fan [...]
Changing the pwm mode can physically damage your hardware.
I would not recommend to do it; the BIOS should set it correctly
as required by the hardware.

Your call, of course.

Guenter

dragonnn, 2015-08-20:
Hmm so why set my bios for fan1 it to 0? An fan control on this doesn't work? I would really wish to take control of all my fans without modification of the hardware.

groeck, 2015-08-20:
> Hmm so why set my bios for fan1 it to 0? An fan control on this doesn't work? I would really wish to take control of all my fans without modification of the hardware.
How would I know ? I would suggest to ask the vendor, or check
the board manual. Maybe the fan connector does not have pwm control
and is expected to be a DC controlled fan, but, again, how would I
know ?

Guenter

dragonnn, 2015-08-20:
Oh, yeah that's true, I had thing all fan connectors are the same and can by easy switched between the pwm and dc mode but according to the manual of my board fan1 and fan2 are only pwm and fan3-5 are dc controlled :/. So I need to split some fan3-5 to have control of all fans in my pc. Thanks for help.

closed 2015-08-20 by dragonnn