groeck/nct6775 #37: Cannot load module in Centos7 after kernel upgrade

unicornis-pl, 2015-12-18:
I freshly compiled the nct6775.ko downloaded from GIT, but the module fails to load:

[root@vbox sensors]# modprobe -v nct6775
insmod /lib/modules/3.10.0-327.3.1.el7.x86_64/nct6775.ko 
modprobe: ERROR: could not insert 'nct6775': Unknown symbol in module, or unknown parameter (see dmesg)

dmesg does not show anything specific:
[root@vbox sensors]# dmesg | grep nct
[    1.310002] ata1.00: supports DRM functions and may not be fully accessible
[    1.310231] ata1.00: supports DRM functions and may not be fully accessible
[    3.389649] xor: automatically using best checksumming function:

It used to work just fine with the kernel 3.10.0-229.7.2.el7.x86_64

Any hints? How can I further debug?

groeck, 2015-12-18:
On 12/18/2015 02:32 PM, unicornis-pl wrote:
> I freshly compiled the nct6775ko downloaded from GIT, but the module fails to load:
>
> [root@vbox sensors]# modprobe -v nct6775
> insmod /lib/modules/3100-32731el7x86_64/nct6775ko
> modprobe: ERROR: could [...]

Unfortunately that information does not include the missing symbol.
There must be something else in the log.

> It used to work just fine with the kernel 3100-22972el7x86_64
>
That doesn't tell me much, unfortunately. It may mean something in the CentOS world,
but nothing to me.

> Any hints? How can I further debug?

Presumably something in the infrastructure was changed, or one of the recent changes
in the nct6775 driver is creating the problem. What is the latest commit in your
copy of the nct6775 source ?

Guenter

unicornis-pl, 2015-12-18:
I dowloaded it today, I guess it is the latest commit c90fb57
Yes, this is definately something changed after a major update I made today (1632 modules updated).

I moved the nct6775.ko file from /usr/lib/modules/3.10.0-327.3.1.el7.x86_64/kernel/drivers/hwmon to /usr/lib/modules/3.10.0-327.3.1.el7.x86_64/ and now it works. 

I do not understand why, but as long as it works....

groeck, 2015-12-18:
> I dowloaded it today, I guess it is the latest commit c90fb57 <https://github.com/groeck/nct6775/commit/c90fb57fb31e9691f5b10ba4bc93ee5b2380ad78>
> Yes, this is definately something changed after a major update I made today (1632 modules updated).
Can you revert the latest commit (c90fb57) and try again ?
Maybe something is wrong with it. If it doesn't help,
you could try reverting f122d853bd as well.

Thanks,
Guenter

groeck, 2015-12-18:
> Can you revert the latest commit (c90fb57) and try again ?
> Maybe something is wrong with it. If it doesn't help,
> you could try reverting f122d853bd as well.
c90fb57 is just the merge, so reverting it won't help. Please try
reverting f122d853bd.

Thanks,
Guenter

unicornis-pl, 2015-12-18:
Most likely it was my stupid error. It worked just fine for all system updates this year but apparently failed with the majir update of today. So far I just copied the nct6775.ko file into hwmon directory of the respective kernel and today it failed.

So I downloaded the zip file from GIT, made the nct6775.ko and copied it, byt I had problems that I described above.

However, running make install (as one should) did the trick.

I can confirm the it works with the lates CenOS 7.

Thanks a lot for your prompt reaction. Issue closed.

closed 2015-12-18 by unicornis-pl