groeck/nct6775 #38: Cannot modify pwm that controls a loud case fan ... what's the trick?

graysky2, 2015-12-21:
Motherboard: MSI MS-7851/H97I AC (MS-7851), BIOS V5.8 08/28/2015
Kernel version: 4.3.3

I have a case fan that runs at full speed despite efforts to slow it in the BIOS.  The controller is on the motherboard and should be adjustable.  Perhaps a buggy BIOS?  I have disabled speed control in the BIOS thinking I can tweak it from Linux.  What I don't understand why the fan speed doesn't change when I change the pwm.

From the [README](https://github.com/groeck/nct6775/blob/master/README), I need to have the pwm1_enable set to manual mode (1) which I did, but despite me modulating the pwm value, the fan speed does not change.  Any thoughts are welcomed.

Here is the freshly booted state:
```
pwm=/sys/devices/platform/nct6775.2576/hwmon/hwmon2/pwm1
speed=/sys/devices/platform/nct6775.2576/hwmon/hwmon2/fan1_input
ctl=/sys/devices/platform/nct6775.2576/hwmon/hwmon2/pwm1_enable

cat $pwm $speed $ctl
255
1142
5
```

Enable manual mode and try a slower speed but no change as you see:

```
echo 1 > $ctl
echo 30 > $pwm

<< I waited 30 sec here to allow the fan to adjust>>

cat $pwm $speed $ctl
30
1142
1
```

groeck, 2015-12-21:
On 12/21/2015 10:22 AM, graysky wrote:
> Motherboard: MSI MS-7851/H97I AC (MS-7851), BIOS V5.8 08/28/2015
> Kernel version: 4.3.3
>
> I have a case fan that runs at full speed despite efforts to slow it in the BIOS. The controller is on the [...]

Did you run the pwmconfig script ? It should give you an idea if there is an association
between pwm control and fan speed.

The next question is if the fan type and the fan connector type match. Are they both 4-pin or
both 3-pin ? If the chassis connector is a 4-pin connector, but the fan is a 3-pin fan, the
motherboard connector might be configured for PWM control but the fan might expect DC voltage
control.

Did you try to connect the chassis fan to a different fan connector on the board ?

FWIW, 1,142 rpm isn't really fast for a chassis fan and should not be noisy. The 4" fans
in my small chassis run at ~1,500 rpm (and low-noise) when idle, and the chassis fan
in my tower chassis runs at ~1,000 rpm when idle.

Guenter

graysky2, 2015-12-22:
>Did you run the pwmconfig script ?

I did run the pwmconfig script and it found no association to/from either of the two.  That is to be expected with with CPU fan controller since it is managed by the BIOS per my settings.

> Are they both 4-pin or both 3-pin ?If the chassis connector is a 4-pin connector, but the fan is a 3-pin fan, the motherboard connector might be configured for PWM control but the fan might expect DC voltage control.

Both connectors on the board are 4-pin but the fan in question is 3-pin.  I will purchase a replacement that is 4-pin to test to see that everything works as expected.

>Did you try to connect the chassis fan to a different fan connector on the board ?

Yes, same result: no change in the fan speed.

I will update this issue when I get a chance to purchase and swap out the fan.  I suspect nothing is wrong with your code and that it is the fan.  Stay tuned and thank you for the thoughtful reply.

groeck, 2015-12-22:
On 12/22/2015 10:15 AM, graysky wrote:
> Did you run the pwmconfig script ?
>
> I did run the pwmconfig script and it found no association to/from either of the two. That is to be expected with with CPU fan controller since it is managed by [...]

I never exclude a code bug, but it is somewhat unlikely in this case as the driver
is widely used.

Check pwmX_mode (DC vs. pwm mode) and compare against the fan specifications.
3-pin fans are usually DC controlled, 4-pin connectors assume PWM control.
The fan might not change speed because it doesn't listen to the pwm signal.

Guenter

graysky2, 2015-12-22:
> Check pwmX_mode (DC vs. pwm mode) and compare against the fan specifications.

I can't find any specs for the fan without a case tear down.  I can say that changing the mode from 1 to 0 and then setting pwm1 to 0 had no effect on the RPMs of this fan.

groeck, 2015-12-22:
On 12/22/2015 12:51 PM, graysky wrote:
>     Check pwmX_mode (DC vs. pwm mode) and compare against the fan specifications.
>
> I can't find any specs for the fan without a case tear down. I can say that changing the mode from 1 to 0 and then setting pwm1 to 0 had no effect on the RPMs of this fan.
>
> —

Question is if it is really pwm1 (and not, say, pwm3). The
association between pwmX and fanX is not necessarily 1:1.

You also have to set fan control (pwmX_enable) to manual.
I assume that you did that ?

Guenter

graysky2, 2015-12-22:
>Question is if it is really pwm1 (and not, say, pwm3). The association between pwmX and fanX is not necessarily 1:1.
>You also have to set fan control (pwmX_enable) to manual. I assume that you did that ?

Yes, no change.
```
echo 0 > /sys/devices/platform/nct6775.2576/hwmon/hwmon2/pwm{1..5}_mode
echo 1 > /sys/devices/platform/nct6775.2576/hwmon/hwmon2/pwm{1..5}_enable
cat /sys/devices/platform/nct6775.2576/hwmon/hwmon2/pwm{1..5}
0
0
0
0
0
cat /sys/devices/platform/nct6775.2576/hwmon/hwmon2/fan{1..5}_input
635
1039
0
0
0
```

groeck, 2015-12-22:
On 12/21/2015 10:22 AM, graysky wrote:
> Motherboard: MSI MS-7851/H97I AC (MS-7851), BIOS V5.8 08/28/2015
> Kernel version: 4.3.3
>
> I have a case fan that runs at full speed despite efforts to slow it in the BIOS. The controller is on the [...]

I did some reading on the board. Looks like you'll need a 4-pin fan.

Your 4-pin fan (the one you talked about in one of your posts) probably
doesn't get below 1000rpm because it is manufactured that way.
CPU fans typically do that, for example, to prevent accidental overheating.

Guenter

graysky2, 2015-12-23:
>I did some reading on the board. Looks like you'll need a 4-pin fan.

Using a 4-pin fan was the solution; I can now access the the full range of RPMs per the specs on the fan.  Thanks for the discussion and for the driver!

closed 2015-12-23 by graysky2