groeck/nct6775 #39: nct6775 not detected on z170 chipset?

andygodber, 2016-03-28:


groeck, 2016-03-28:
Support for NCT6791D was added to the driver after v3.10. I guess Centos/Redhat didn't back-port the updated driver. Did you try building and installing the driver from here ?

pjperry, 2016-03-29:
elrepo.org have built this driver as a kmod package for RHEL/CentOS (kmod-nct6775) which you could try:

http://elrepo.org/linux/elrepo/el7/x86_64/RPMS/kmod-nct6775-0.0-3.20151222git.el7.elrepo.x86_64.rpm

andygodber, 2016-03-29:
Thanks both - despite having thought I'd built from the GIT here, I must have messed up somewhere. So I retraced all my steps, deleted various copies of nct6775.ko from different location, and rebuilt from here - all worked ok.

Phil - I tried booting from the kernel (centos.plus) the repo from your suggestion created. Unfortunately this didnt work, leaving the same message 'no such device' from modprobe. Anything I can do to try and fix/help that?

Will close this thread later.

closed 2016-03-29 by andygodber