groeck/nct6775 #40: [NCT6776F] Issue with CPU FAN speed

pierretom, 2016-04-03:
Hello,

CPU FAN speed appears only if BIOS was previously entered and the "H/W Monitor" 
page has been accessed.
Its set to "Automatic mod" in the BIOS, same happen on "Full On".

Without BIOS access:
```
nct6776-isa-0290
Adapter: ISA adapter
Vcore:          +0.90 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:           +3.31 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:          +3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:            +1.35 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:            +1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:            +0.62 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:           +3.44 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:           +3.36 V  (min =  +2.70 V, max =  +3.63 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:             0 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
SYSTIN:         +28.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:         +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN:         +15.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
PCH_CHIP_TEMP:   +0.0°C  
PCH_CPU_TEMP:    +0.0°C  
PCH_MCH_TEMP:    +0.0°C  
intrusion0:    ALARM
intrusion1:    ALARM
beep_enable:   disabled
```

After BIOS access:
```
nct6776-isa-0290
Adapter: ISA adapter
Vcore:          +0.90 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:           +3.31 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:          +3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:            +1.43 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:            +1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:            +0.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:           +3.44 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:           +3.36 V  (min =  +2.70 V, max =  +3.63 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:           982 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
SYSTIN:         +34.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:         +29.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN:          +5.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
PCH_CHIP_TEMP:   +0.0°C  
PCH_CPU_TEMP:    +0.0°C  
PCH_MCH_TEMP:    +0.0°C  
intrusion0:    ALARM
intrusion1:    ALARM
beep_enable:   disabled
```
Any ideas ?

groeck, 2016-04-07:
What board is this ?

pierretom, 2016-04-07:
> What board is this ?

ASRock 990FX Extreme9

With the ASRock Extreme Tuning Utility, I can see the CPU fan speed without BIOS access.
I also found [this file](http://test01.atspace.eu/990FXE9.xml) that come with it.

groeck, 2016-04-07:
https://github.com/groeck/nct6775/blob/master/README.ASRock

pierretom, 2016-04-08:
Works with
```
isaset -y -f 0x2e 0x87
Data mismatch, wrote 0x87, read 0xff back.
isaset -y -f 0x2e 0x87
Data mismatch, wrote 0x87, read 0x27 back.
isaset -y 0x2e 0x2f 0x7 0x8
isaset -y 0x2e 0x2f 0xf0 0xb2
isaset -y 0x2e 0x2f 0xf1 0xb3
```

Thanks groeck.

closed 2016-04-08 by pierretom