groeck/nct6775 #41: Many empty values with NCT6791D on Asus H110M-C

kapsh, 2016-05-22:
I have the H110M-C motherboard and `sensors-detect` detects my sensor as: 
```
Driver `nct6775' (built-in):
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6791D Super IO Sensors' (confidence: 9)
```
But there are many "inX" values is absent in sensors3.conf and `sensors` shows many false positives:
```
nct6791-isa-0290
Adapter: ISA adapter
Vcore:                  +0.39 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.44 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.44 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +2.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.99 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.38 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.10 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.99 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.99 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                  1081 RPM  (min =    0 RPM)
fan2:                  1192 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan6:                     0 RPM
SYSTIN:                 +26.0°C  (high = +98.0°C, hyst = +95.0°C)  sensor = thermistor
CPUTIN:                 +33.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +26.5°C    sensor = thermistor
AUXTIN1:                +26.0°C    sensor = thermistor
AUXTIN2:                +26.0°C    sensor = thermistor
AUXTIN3:                +95.0°C    sensor = thermistor
PECI Agent 0:           +30.0°C  (high = +98.0°C, hyst = +95.0°C)
                                 (crit = +100.0°C)
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled
```
I didn't found my m/b on page http://www.lm-sensors.org/wiki/Configurations/Asus (and this wiki seems to be down now).

groeck, 2016-05-22:
On 05/22/2016 02:34 PM, Kapshuna Alexander wrote:
> I have the H110M-C motherboard and |sensors-detect| detects my sensor as:
>
> |Driver `nct6775' (built-in): * ISA bus, address 0x290 Chip `Nuvoton NCT6791D Super IO Sensors' (confidence: 9) |
>
> But there are many "inX" values is absent in sensors3.conf and |sensors| shows many false positives:
>
The inX values are not really absent - the limits are not configured. This is causing the false positives.

> |nct6791-isa-0290 Adapter: ISA adapter Vcore: +0.39 V (min = +0.00 V, max = +1.74 V) in1: +1.01 V (min = +0.00 V, max = +0.00 V) ALARM AVCC: +3.44 V (min = +2.98 V, max = +3.63 V) +3.3V: +3.44 V [...]

> = thermistor AUXTIN0: +26.5°C sensor = thermistor AUXTIN1: +26.0°C sensor = thermistor AUXTIN2: +26.0°C sensor = thermistor AUXTIN3: +95.0°C sensor = thermistor PECI Agent 0: +30.0°C (high = [...]
Yes, unfortunately it is down. Someone suggested to move the configurations to github,
which would make a lot of sense.

Guenter

kapsh, 2016-05-26:
I see, thanks. So, how can I configure limits for myself?

groeck, 2016-07-06:
Set the limits in /etc/sensors3.conf, then run "sudo sensors -s"

closed 2016-07-06 by groeck