groeck/nct6775 #42: NCT6793D with ID 0xd121 on Asus Z170-Deluxe

h3po, 2016-07-06:
When i force_id=0xd120 on my Asus Z170-Deluxe, the sensors seem to be reporting correctly. I haven't tried fan controlling yet. There's one issue, sensors finds a second NCT6793D bogus adapter that reports only garbage:

	acpitz-virtual-0
	Adapter: Virtual device
	temp1:        +27.8°C  (crit = +119.0°C)
	temp2:        +29.8°C  (crit = +119.0°C)

	nct6793-isa-0290
	Adapter: ISA adapter
	in0:                       +0.40 V  (min =  +0.00 V, max =  +1.74 V)
	in1:                       +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in2:                       +3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in3:                       +3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in4:                       +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in5:                       +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in6:                       +0.00 V  (min =  +0.00 V, max =  +0.00 V)
	in7:                       +3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in8:                       +3.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in9:                       +1.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in10:                      +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in11:                      +1.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in12:                      +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in13:                      +0.59 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	in14:                      +1.13 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
	fan1:                        0 RPM  (min =    0 RPM)
	fan2:                      739 RPM  (min =    0 RPM)
	fan3:                      651 RPM  (min =    0 RPM)
	fan4:                      459 RPM  (min =    0 RPM)
	fan5:                      626 RPM  (min =    0 RPM)
	fan6:                        0 RPM
	SYSTIN:                    +28.0°C  (high = +98.0°C, hyst = +95.0°C)  sensor = thermistor
	CPUTIN:                    +29.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
	AUXTIN0:                  +122.0°C    sensor = thermistor
	AUXTIN1:                   +45.0°C    sensor = thermistor
	AUXTIN2:                   +15.0°C    sensor = thermistor
	AUXTIN3:                   +19.0°C    sensor = thermistor
	PECI Agent 0:              +27.0°C  (high = +98.0°C, hyst = +95.0°C)
		                            (crit = +100.0°C)
	PECI Agent 0 Calibration:  +29.5°C  
	PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
	PCH_CHIP_TEMP:              +0.0°C  
	intrusion0:               ALARM
	intrusion1:               ALARM
	beep_enable:              disabled

	asus-isa-0000
	Adapter: ISA adapter
	cpu_fan:        0 RPM

	nct6793-isa-fff8
	Adapter: ISA adapter
	in0:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in1:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in2:                    +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
	in3:                    +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
	in4:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in5:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in6:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in7:                    +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
	in8:                    +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
	in9:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in10:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in11:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in12:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in13:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	in14:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
	fan1:                     0 RPM  (min =    0 RPM)  ALARM
	fan2:                     0 RPM  (min =    0 RPM)  ALARM
	fan6:                     0 RPM  ALARM
	SYSTIN:                  -1.0°C  (high =  -1.0°C, hyst =  -1.0°C)  sensor = CPU diode
	CPUTIN:                  -1.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = CPU diode
	AUXTIN0:                 -1.0°C    sensor = CPU diode
	AUXTIN1:                 -1.0°C    sensor = CPU diode
	AUXTIN2:                 -1.0°C    sensor = CPU diode
	AUXTIN3:                 -1.0°C    sensor = CPU diode
	Virtual_TEMP:            -1.0°C  (high =  -1.0°C, hyst =  -1.0°C)  ALARM
	PCH_CHIP_CPU_MAX_TEMP:   -1.0°C  
	PCH_CHIP_TEMP:           -1.0°C  
	PCH_CPU_TEMP:            +0.0°C  
	intrusion0:            ALARM
	intrusion1:            ALARM
	beep_enable:           enabled

fan6 either does not exist or is CPU_OPT. The board has a header for an external temperature sensor (to be placed at the memory banks) which is probably AUXTIN0. AUXTIN2, AUXTIN3 must be garbage, since I'm at 21°C ambient.
Please let me know if I can provide any more info to facilitate integrating the new device into the driver. I might try it myself, but I'm dealing with all kinds of hardware issues atm. Happy to see the temperature reading working. Thanks a lot for your work!

groeck, 2016-07-06:
The NCT6793D should not require force_id, and the second chip may possibly show up because of it (after all, it forces the ID of anything to NCT6793D).

What is the output of sensors-detect ?

h3po, 2016-07-06:
well it finds the chip, but treats it as unknown because of the ID 0xd121.
```
Found unknown chip with ID 0xd121
        (logical device B has address 0x290, could be sensors)
```
i can't see the chip on my board as it is now in the case, but i think it is an NCT6793D because of this picture from a review: https://www.techpowerup.com/reviews/ASUS/Z170-DELUXE/images/super_io.jpg
[sensors-detect.txt](https://github.com/groeck/nct6775/files/350814/sensors-detect.txt)

groeck, 2016-07-06:
The driver masks off the lowest 4 bits before matching, and should
therefore match against 0xd120.

It seems that sensors-detect matches a wrong value, though, since it
checks for 0xD11x instead of 0xD12x.

If you don't use force_id, what do you see in the kernel log ?

h3po, 2016-07-06:
oh. now it shows up without forcing the id as well. then the problem was the acpi address conflict with the bios?

```
╭─h3po@daenerys  ~  
╰─$ sudo modprobe nct6775                                                                                           
╭─h3po@daenerys  ~  
╰─$ dmesg | tail 
...
[   85.398602] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
[   85.398606] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM) (20160108/utaddress-255)
[   85.398609] ACPI: This conflict may cause random problems and system instability
[   85.398609] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
```

closed 2016-07-06 by h3po

groeck, 2016-07-06:
Correct.

The second device showed up because the driver checks for two chips, and forced
the ID of the non-existing second chip to 0xd121 as requested.

jcharaoui, 2016-08-04:
I have the exact same problem as the OP, but with a different motherboard: ASUS Z170 PRO GAMING. The kernel reports these messages when inserting the module:

    [18167.695259] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290
    [18167.695272] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM) (20160108/utaddress-255)
    [18167.695284] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
    [18167.695355] nct6775: Enabling hardware monitor logical device mappings.
    [18167.695386] nct6775: Found NCT6793D or compatible chip at 0x4e:0xfff8`

However in my case the `sensors` command only report the bogus values from device `nct6793-isa-fff8`. Is this also an issue strictly with lm-sensors?

jcharaoui, 2016-08-04:
I just went reviewed the lm-sensors code repository and saw that [NCT6793D support will be part of the next release](https://github.com/groeck/lm-sensors/blob/e39dbd3e5609ede626710d2811025c0c97345088/CHANGES#L9).

groeck, 2016-08-04:
Again, the bogus chip shows up because force_id was specified. I think I'll restrict force_id to only work if the ID read from the chip is not 0xffff.

In the case of ASUS Z170 PRO GAMING, the first chip does not show up because ACPI reserves the memory space for itself. This requires boot with acpi_enforce_resources=lax in the command line (and accepting the associated risk).

vilagithub, 2016-09-28:
Same issue here (no detection of NCT6793D with id 0xd121) with a Z170-deluxe motherboard running ubuntu xenial 16.04 LTS
'acpi_enforce_resources=lax' felt too risky after reading https://web.archive.org/web/20140630034601/http://www.lm-sensors.org/wiki/FAQ/Chapter3#Mysensorshavestoppedworkinginkernel2.6.31 

What did work was to add in /etc/modules:
nct6775 force_id=0xd120

Any chance the mainline kernel can recognize the chip properly out of the box ?

vilagithub, 2016-09-28:
Hmpf, not sure how I tested previously >-/

'acpi_enforce_resources=lax' in /etc/default/grub and 'nct6775' in /etc/modules is the only working combination.

So, I'd feel better *without* 'acpi_enforce_resources=lax' , is there a solution for that ? 
It's the first time I encounter such issues so any lecturing welcome ;-)

groeck, 2016-09-28:
On Wed, Sep 28, 2016 at 09:17:44AM -0700, vilagithub wrote:
> Same issue here (no detection of NCT6793D with id 0xd121) with a Z170-deluxe motherboard running ubuntu xenial 16.04 LTS
> 'acpi_enforce_resources=lax' felt too risky after reading [...]

The kernel does recognize the chip. That is not the problem. The problem is
that the BIOS wants to have sole access to the chip.

Apparently you found a way of bypassing the warning. Removing the message
from the driver, ie no longer trying to inform you about the risk, would
accomplish the same.

Really, it is your call. If you think that swimming is safer after removing
the "Warning - Alligators" sign, go ahead. Just keep in mind that the alligators
are still there.

Thanks,
Guenter

vilagithub, 2016-09-30:
@groeck: Sorry, see my later comment.

I *don't* feel safe bypassing the warning *at all*.

I didn't bypass the warning, not sure how I ended up thinking I could, but I couldn't reproduce, so operator error.

I would really prefer a solution where I don't have to use 'acpi_enforce_resources=lax'  which is *required* so far.

I did activate it to monitor a new setup behaving strangely (bad default RAM bios settings was causing issue apparently, fixed now) and I would prefer to keep this monitoring but I will probably just remove it instead once I'm happy with the hardware behaviour.

So, trying to re-formulate: what are my options to get monitoring ? 

- nct6775 provides exactly the data I'm after but alligators...
- acpi itself... I couldn't find any working setup (but I don't really know where/how to search)

groeck, 2016-10-02:
You have a number of options I am aware of:
- do not use this driver
- live with the warning
- analyze the ACPI DSDT (Differentiated System Description Table) and write a driver using ACPI to monitor the system
- Ask ASUS for assistance

ASUS does not support Linux on their desktop boards (as mentioned in README), so asking them for assistance would likely be futile. Analyzing the DSDT may reveal that ACPI does not expose the data you are looking for, so that option may not be feasible either. Effectively this means that you'll either have to live with the risk or not use the driver.

vilagithub, 2016-10-04:
Thanks for the comprehensive answer.

I didn't know ASUS wasn't linux-friendly :-/ Shame.