groeck/nct6775 #44: Asus Fan Extension Card support?

ulteq, 2016-09-09:
I'm using an Asus X99 A-II motherboard and everything works great out of the box. Including the `T_Sensor1` temperature reading.

This is how my sensors output looks like: https://ptpb.pw/l2aE.output

All fans are properly detected and the `T_Sensor1` temperature shows up as `Virtual_TEMP`.

However, none of the additional sensor and fan headers on the fan extension card are detected by default.

Here are some uefi screenshots:

[image]
[image]
[image]

The following steps where necessary to get access to both the `T_Sensor1` and the `EXT_Sensor1` reading:

- Select [T_Sensor1 as Chassis Fan 1 Q-Fan Source](https://cloud.githubusercontent.com/assets/9437207/18383456/ff34469c-7685-11e6-8fe2-ce66f4ebd429.png)
- Select [EXT_Sensor1 as Chassis Fan 2 Q-Fan Source](https://cloud.githubusercontent.com/assets/9437207/18383462/022d96be-7686-11e6-9bd7-5f0946dc203e.png)
- Remove the `(mask & (1 << src))` condition from this line: https://github.com/groeck/nct6775/blob/master/nct6775.c#L3962

This is how my sensors output looks like after the change: https://ptpb.pw/JsuR.output
The 1st `Virtual Temp` reading shows the `T_Sensor1` reading.
The 2nd `Virtual Temp` reading shows the `EXT_Sensor1` reading.

Is there something I can do to help you to support the fan expansion card?

groeck, 2016-09-09:
> I'm using an Asus X99 A-II motherboard and everything works great out of the box. Including the |T_Sensor1| temperature reading.
>
> This is how my sensors output looks like: https://ptpb.pw/l2aE.output
>
> All fans are properly detected and the |T_Sensor1| temperature shows up as |Virtual_TEMP|.
>
> However, none of the additional sensor and fan headers on the fan extension card are detected by default.
>
> Here are some uefi screenshots:
>
> ext_sensor1 <https://cloud.githubusercontent.com/assets/9437207/18383183/9e8407a2-7684-11e6-9b70-fae8cacc8a4c.png>
> ext_fans <https://cloud.githubusercontent.com/assets/9437207/18383185/a1eececc-7684-11e6-9de2-2831e35357c7.png>
> ext_fan1 <https://cloud.githubusercontent.com/assets/9437207/18383189/a3e2d9b2-7684-11e6-820f-2c72df356922.png>
>
> The following steps where necessary to get access to both the |T_Sensor1| and the |EXT_Sensor1| reading:
>
>   * Select T_Sensor1 as Chassis Fan 1 Q-Fan Source <https://cloud.githubusercontent.com/assets/9437207/18383456/ff34469c-7685-11e6-8fe2-ce66f4ebd429.png>
>   * Select EXT_Sensor1 as Chassis Fan 2 Q-Fan Source <https://cloud.githubusercontent.com/assets/9437207/18383462/022d96be-7686-11e6-9bd7-5f0946dc203e.png>
>   * Remove the |(mask & (1 << src))| condition from this line: https://github.com/groeck/nct6775/blob/master/nct6775.c#L3962
>
> This is how my sensors output looks like after the change: https://ptpb.pw/JsuR.output
> The 1st |Virtual Temp| reading shows the |T_Sensor1| reading.
> The 2nd |Virtual Temp| reading shows the |EXT_Sensor1| reading.
Virtual_TEMP:           +28.0°C
PECI Agent 0:           +31.0°C
Virtual_TEMP:           +29.0°C
PECI Agent 0:           +31.0°C

PECI Agent 0 is for sure a duplicate. Virtual_TEMP is interesting; I thought it would
reflect a single register written to by software, but maybe not. I'll look into the
datasheet some more.

Guenter

ulteq, 2016-09-11:
> PECI Agent 0 is for sure a duplicate.

The duplicate `PECI Agent 0` entries show different temperatures under heavy system load: https://ptpb.pw/ncXt

groeck, 2016-09-11:
On 09/11/2016 09:03 AM, ulteq wrote:
>     PECI Agent 0 is for sure a duplicate.
>
> The duplicate |PECI Agent 0| entries show different temperatures under heavy system load: https://ptpb.pw/ncXt
>

No idea how this works. Either the temperatures fluctuate quickly between reads, or the PECI
sources are somehow virtualized as well. I am not that concerned about it, since coretemp
will report those temperatures as well.

Can you try the tip of the tree and let me know if it reports all virtual temperatures ?

Thanks,
Guenter

ulteq, 2016-09-11:
> Can you try the tip of the tree and let me know if it reports all virtual temperatures ?

It does report both now: https://ptpb.pw/tyO5

groeck, 2016-10-02:
Re pwm controls: Those should be available if the chip is programmed correctly by the BIOS. How do the raw sysfs attributes look like (in /sys/class/hwmon/hwmonX) ?

ulteq, 2016-10-11:
> How do the raw sysfs attributes look like

```
[ulteq@arch hwmon2]$ pwd
/sys/class/hwmon/hwmon2
[ulteq@arch hwmon2]$ ls
beep_enable     fan4_min        in11_alarm  in2_beep   in7_input              pwm1_auto_point4_pwm      pwm2_crit_temp_tolerance    pwm3_auto_point5_pwm      pwm4_floor                pwm5_step_up_time         pwm6_temp_tolerance  temp5_input
device          fan4_pulses     in11_beep   in2_input  in7_max                pwm1_auto_point4_temp     pwm2_enable                 pwm3_auto_point5_temp     pwm4_mode                 pwm5_stop_time            subsystem            temp5_label
fan1_alarm      fan4_target     in11_input  in2_max    in7_min                pwm1_auto_point5_pwm      pwm2_floor                  pwm3_crit_temp_tolerance  pwm4_start                pwm5_target_temp          temp10_input         temp5_offset
fan1_beep       fan4_tolerance  in11_max    in2_min    in8_alarm              pwm1_auto_point5_temp     pwm2_mode                   pwm3_enable               pwm4_step_down_time       pwm5_temp_sel             temp10_label         temp5_type
fan1_input      fan5_alarm      in11_min    in3_alarm  in8_beep               pwm1_crit_temp_tolerance  pwm2_start                  pwm3_floor                pwm4_step_up_time         pwm5_temp_tolerance       temp1_input          temp6_input
fan1_min        fan5_beep       in12_alarm  in3_beep   in8_input              pwm1_enable               pwm2_step_down_time         pwm3_mode                 pwm4_stop_time            pwm6                      temp1_label          temp6_label
fan1_pulses     fan5_input      in12_beep   in3_input  in8_max                pwm1_floor                pwm2_step_up_time           pwm3_start                pwm4_target_temp          pwm6_auto_point1_pwm      temp1_max            temp6_offset
fan1_target     fan5_min        in12_input  in3_max    in8_min                pwm1_mode                 pwm2_stop_time              pwm3_step_down_time       pwm4_temp_sel             pwm6_auto_point1_temp     temp1_max_hyst       temp6_type
fan1_tolerance  fan5_pulses     in12_max    in3_min    in9_alarm              pwm1_start                pwm2_target_temp            pwm3_step_up_time         pwm4_temp_tolerance       pwm6_auto_point2_pwm      temp1_offset         temp7_alarm
fan2_alarm      fan5_target     in12_min    in4_alarm  in9_beep               pwm1_step_down_time       pwm2_temp_sel               pwm3_stop_time            pwm5                      pwm6_auto_point2_temp     temp1_type           temp7_beep
fan2_beep       fan5_tolerance  in13_alarm  in4_beep   in9_input              pwm1_step_up_time         pwm2_temp_tolerance         pwm3_target_temp          pwm5_auto_point1_pwm      pwm6_auto_point3_pwm      temp2_alarm          temp7_crit
fan2_input      fan6_alarm      in13_beep   in4_input  in9_max                pwm1_stop_time            pwm2_weight_duty_base       pwm3_temp_sel             pwm5_auto_point1_temp     pwm6_auto_point3_temp     temp2_beep           temp7_input
fan2_min        fan6_input      in13_input  in4_max    in9_min                pwm1_target_temp          pwm2_weight_duty_step       pwm3_temp_tolerance       pwm5_auto_point2_pwm      pwm6_auto_point4_pwm      temp2_input          temp7_label
fan2_pulses     fan6_pulses     in13_max    in4_min    intrusion0_alarm       pwm1_temp_sel             pwm2_weight_temp_sel        pwm4                      pwm5_auto_point2_temp     pwm6_auto_point4_temp     temp2_label          temp7_max
fan2_target     fan6_target     in13_min    in5_alarm  intrusion0_beep        pwm1_temp_tolerance       pwm2_weight_temp_step       pwm4_auto_point1_pwm      pwm5_auto_point3_pwm      pwm6_auto_point5_pwm      temp2_max            temp7_max_hyst
fan2_tolerance  fan6_tolerance  in14_alarm  in5_beep   intrusion1_alarm       pwm2                      pwm2_weight_temp_step_base  pwm4_auto_point1_temp     pwm5_auto_point3_temp     pwm6_auto_point5_temp     temp2_max_hyst       temp8_input
fan3_alarm      in0_alarm       in14_beep   in5_input  intrusion1_beep        pwm2_auto_point1_pwm      pwm2_weight_temp_step_tol   pwm4_auto_point2_pwm      pwm5_auto_point4_pwm      pwm6_crit_temp_tolerance  temp2_offset         temp8_label
fan3_beep       in0_beep        in14_input  in5_max    name                   pwm2_auto_point1_temp     pwm3                        pwm4_auto_point2_temp     pwm5_auto_point4_temp     pwm6_enable               temp2_type           temp9_input
fan3_input      in0_input       in14_max    in5_min    power                  pwm2_auto_point2_pwm      pwm3_auto_point1_pwm        pwm4_auto_point3_pwm      pwm5_auto_point5_pwm      pwm6_floor                temp3_input          temp9_label
fan3_min        in0_max         in14_min    in6_alarm  pwm1                   pwm2_auto_point2_temp     pwm3_auto_point1_temp       pwm4_auto_point3_temp     pwm5_auto_point5_temp     pwm6_mode                 temp3_label          uevent
fan3_pulses     in0_min         in1_alarm   in6_beep   pwm1_auto_point1_pwm   pwm2_auto_point3_pwm      pwm3_auto_point2_pwm        pwm4_auto_point4_pwm      pwm5_crit_temp_tolerance  pwm6_start                temp3_offset
fan3_target     in10_alarm      in1_beep    in6_input  pwm1_auto_point1_temp  pwm2_auto_point3_temp     pwm3_auto_point2_temp       pwm4_auto_point4_temp     pwm5_enable               pwm6_step_down_time       temp3_type
[... cut ...]
fan4_alarm      in10_input      in1_max     in6_min    pwm1_auto_point2_temp  pwm2_auto_point4_temp     pwm3_auto_point3_temp       pwm4_auto_point5_temp     pwm5_mode                 pwm6_stop_time            temp4_label
fan4_beep       in10_max        in1_min     in7_alarm  pwm1_auto_point3_pwm   pwm2_auto_point5_pwm      pwm3_auto_point4_pwm        pwm4_crit_temp_tolerance  pwm5_start                pwm6_target_temp          temp4_offset
fan4_input      in10_min        in2_alarm   in7_beep   pwm1_auto_point3_temp  pwm2_auto_point5_temp     pwm3_auto_point4_temp       pwm4_enable               pwm5_step_down_time       pwm6_temp_sel             temp4_type
```

groeck, 2016-11-12:
I finally had some time to look into this. The fan extension board has its own fan controller, a Nuvoton NCT7802Y. The Linux kernel has a driver for this chip, or you can use https://github.com/groeck/nct7802.
The chip uses I2C to communicate to the CPU. You would have to find the i2c bus and i2c address used by the chip. Try running sensors-detect from the lm-sensors package and see if it finds the chip.

closed 2016-11-12 by groeck