groeck/nct6775 #46: NCT6776D on Supermicro X11SAE-F no fan RPM readings

Aikhjarto, 2016-11-19:
Hi,

I have a Supermicro X11SAE-F which has NCT6776D. Seems that the fan RPM reading is missing.
```
# sensors
nct6776-isa-0a30
Adapter: ISA adapter
Vcore:          +0.60 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:           +3.34 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:          +3.34 V  (min =  +2.98 V, max =  +3.63 V)
in4:            +1.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:            +1.25 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:            +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:           +3.34 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:           +2.98 V  (min =  +2.70 V, max =  +3.63 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:             0 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
SYSTIN:         +37.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:        +123.5°C  (high = +80.0°C, hyst = +75.0°C)  ALARM  sensor = CPU diode
AUXTIN:         +26.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
PCH_CHIP_TEMP:   +0.0°C
PCH_CPU_TEMP:    +0.0°C
PCH_MCH_TEMP:    +0.0°C
intrusion0:    ALARM
intrusion1:    ALARM
beep_enable:   disabled
```
Has someone a idea to get the fan RPM readings?

groeck, 2016-11-19:
Odd. Can you run sensors-detect ? Maybe the system has a second controller for fan control. Unlikely, but it would not be the first time.

Aikhjarto, 2016-11-20:
Here is the ouput of sensors-detect
```
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Supermicro Super Server [0123456789]
# Board: Supermicro X11SAE-F
# Kernel: 4.8.8-1.g7b7d4ed-default x86_64
# Processor: Intel(R) Xeon(R) CPU E3-1245 v5 @ 3.50GHz (6/94/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT5573D/NCT5577D/NCT6776F Super IO Sensors' Success!
    (address 0xa30, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no):
Found `IPMI BMC KCS' at 0xca2...                            Success!
    (confidence 8, driver `to-be-written')

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
[... 89 lines cut ...]
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: DPDDC-B (i2c-5)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: DPDDC-D (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes

Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `nct6775':
  * ISA bus, address 0xa30
    Chip `Nuvoton NCT5573D/NCT5577D/NCT6776F Super IO Sensors' (confidence: 9)

Driver `to-be-written':
  * ISA bus, address 0xca2
    Chip `IPMI BMC KCS' (confidence: 8)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Note: there is no driver for IPMI BMC KCS yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.
```

Aikhjarto, 2016-11-26:
> Maybe the system has a second controller for fan control.

I open the chassis and checked manually if there are possibly any other fan controll chips. However, I didn't find any.

Aikhjarto, 2016-12-17:
People with similar problems found out that setting the kernel parameter acpi_enforce_resources=lax can help with not-working sensor readings. However, at least on my X11SAE-F running kernel 4.9 this doesn't do anything.

groeck, 2016-12-17:
acpi_enforce_resources only works if the driver does not load at all, and then only to some degree. If the kernel reports a resource conflict, there is a danger that the BIOS or the BMC and the driver access the Super-IO chip at the same time, which can result in system crashes. So use it on your own risk.
For X11SAE-F, the board manual clearly suggests that the fan speed is monitored and controlled by the BMC. It might be possible to access its readings through some ipmi tool, but I don't know what is available and supported.

closed 2016-12-17 by groeck