groeck/nct6775 #47: No readings on "NCT6791D-A"

kevinlekiller, 2016-11-24:
Hello, I have a motherboard with a "NCT6791D-A", it is detected as a "NCT6791D" by sensors-detect (lm_sensors 3.4.0-2, kernel 4.8.10):

    Now follows a summary of the probes I have just done.

    Driver `nct6775':
      * ISA bus, address 0x290
        Chip `Nuvoton NCT6791D Super IO Sensors' (confidence: 9)

    Driver `coretemp':
      * Chip `Intel digital thermal sensor' (confidence: 9)

The output by sensors:

    acpitz-virtual-0
    Adapter: Virtual device
    temp1:        +27.8°C  (crit = +105.0°C)
    temp2:        +29.8°C  (crit = +105.0°C)

    coretemp-isa-0000
    Adapter: ISA adapter
    Physical id 0:  +35.0°C  (high = +80.0°C, crit = +100.0°C)
    Core 0:         +35.0°C  (high = +80.0°C, crit = +100.0°C)
    Core 1:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
    Core 2:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
    Core 3:         +30.0°C  (high = +80.0°C, crit = +100.0°C)

    asus-isa-0000
    Adapter: ISA adapter
    cpu_fan:        0 RPM

On another motherboard with a NCT6791D (without the -A), the output of sensors shows all the information on voltages/fan speeds/temps/etc.

I tried looking up information on NCT6791D-A but could not find anything, here's an image of the chip (which I know doesn't help, but at least confirms the -A variant exists, since it's not listed on the Nuvoton website as far as I can see): https://i.imgur.com/zcWHMyf.jpg

kevinlekiller, 2016-11-24:
I think this is the same issue as https://github.com/groeck/lm-sensors/issues/10

Since the Sabertooth X99 uses the same "NCT6791D-A" chip according to a picture in this review:

https://www.hardwareluxx.ru/index.php/artikel/hardware/mainboards/34461-asus-sabertooth-x99-test.html?start=3

This picture of the chip: http://www.hardwareluxx.de/media/jphoto/artikel-galerien/asus-sabertooth-x99/img-20-950x633.jpg

groeck, 2016-11-24:
Correct. dmesg will most likely show that there is a resource conflict.

kevinlekiller, 2016-11-24:
Yep, same issue it seems (output of dmesg):

    [    1.845997] nct6775: Enabling hardware monitor logical device mappings.
    [    1.846010] nct6775: Found NCT6791D or compatible chip at 0x2e:0x290
    [    1.846014] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM_) (20150930/utaddress-254)
    [    1.846017] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

Will test with `acpi_enfore_resources=lax` like you mentioned in the other issue, thanks!

kevinlekiller, 2016-11-24:
No difference with `acpi_enfore_resources=lax` unfortunately:

    [    0.000000] Kernel command line: BOOT_IMAGE=/vmlinuz-linux-lts root=UUID=ba07f51c-c943-40d6-91c6-3462a9c53637 rw quiet intel_iommu=on drm_kms_helper.edid_firmware=HDMI-A-2:edid/edidhdmi2.bin acpi_enfore_resources=lax
    [    1.961010] nct6775: Enabling hardware monitor logical device mappings.
    [    1.961020] nct6775: Found NCT6791D or compatible chip at 0x2e:0x290
    [    1.961023] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000290-0x0000000000000299 (\_GPE.HWM_) (20150930/utaddress-254)
    [    1.961025] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

kevinlekiller, 2016-11-24:
Disregard those last 2 comments, I copy pasted `acpi_enfore_resources=lax` from https://github.com/groeck/lm-sensors/issues/10#issuecomment-248109978 but there's a typo (missing a c).

Fixed the typo and now I can see all the sensor information:

```
nct6791-isa-0290
Adapter: ISA adapter
Vcore:                  +0.90 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.41 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.41 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +2.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.72 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.47 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.41 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +2.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in13:                   +0.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   310 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                   336 RPM
SYSTIN:                 +29.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +32.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +40.0°C    sensor = thermistor
AUXTIN1:                -62.0°C    sensor = thermistor
AUXTIN2:                +35.0°C    sensor = thermistor
AUXTIN3:               -128.0°C    sensor = thermistor
PECI Agent 0:           +32.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

Thanks for the assist!

closed 2016-11-24 by kevinlekiller