groeck/nct6775 #48: Fans not displayed in `sensors` output

hgregorian, 2016-12-19:
Z10PE-D16 with NCT6779D (7 Noctua PWM fans) 

sensors output:
```
[root@thoth ~]# sensors-detect --auto && sensors
# sensors-detect revision 3.4.0-4 (2016-06-01)
# System: ASUSTeK COMPUTER INC. Z10PE-D16 Series
# Kernel: 3.10.0-514.2.2.el7.x86_64 x86_64
# Processor: Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz (6/79/1)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT5532D/NCT6779D Super IO Sensors'          Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Found `IPMI BMC KCS' at 0xca2...                            Success!
    (confidence 8, driver `to-be-written')

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): 

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:8d22 at 0000:00:1f.3.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

Next adapter: AST i2c bit bus (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: igb BB (i2c-1)
Do you want to scan it? (YES/no/selectively): 

Next adapter: igb BB (i2c-2)
Do you want to scan it? (YES/no/selectively): 

Next adapter: SMBus I801 adapter at 0580 (i2c-3)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x48
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `SMSC EMC1023'...                               No
Probing for `SMSC EMC1043'...                               No
Probing for `SMSC EMC1053'...                               No
Probing for `SMSC EMC1063'...                               No
Client found at address 0x58
Probing for `Analog Devices ADT7462'...                     No
Probing for `Andigilog aSC7512'...                          No

Now follows a summary of the probes I have just done.

Driver `to-be-written':
  * ISA bus, address 0xca2
    Chip `IPMI BMC KCS' (confidence: 8)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT5532D/NCT6779D Super IO Sensors' (confidence: 9)

Note: there is no driver for IPMI BMC KCS yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Unloading i2c-dev... OK

i350bb-pci-0300
Adapter: PCI adapter
loc1:         +63.0°C  (high = +120.0°C, crit = +110.0°C)

power_meter-acpi-0
Adapter: ACPI interface
power1:        4.29 MW (interval =   1.00 s)

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +39.0°C  (high = +80.0°C, crit = +90.0°C)
Core 0:         +31.0°C  (high = +80.0°C, crit = +90.0°C)
Core 1:         +33.0°C  (high = +80.0°C, crit = +90.0°C)
Core 2:         +31.0°C  (high = +80.0°C, crit = +90.0°C)
Core 3:         +32.0°C  (high = +80.0°C, crit = +90.0°C)
Core 4:         +31.0°C  (high = +80.0°C, crit = +90.0°C)
Core 5:         +32.0°C  (high = +80.0°C, crit = +90.0°C)
Core 6:         +32.0°C  (high = +80.0°C, crit = +90.0°C)
Core 7:         +32.0°C  (high = +80.0°C, crit = +90.0°C)

coretemp-isa-0001
Adapter: ISA adapter
Physical id 1:  +42.0°C  (high = +80.0°C, crit = +90.0°C)
Core 0:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 1:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 2:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 3:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 4:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 5:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 6:         +34.0°C  (high = +80.0°C, crit = +90.0°C)
Core 7:         +33.0°C  (high = +80.0°C, crit = +90.0°C)

nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +2.03 V  (min =  +0.00 V, max =  +1.74 V)  ALARM
in1:                    +0.21 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.26 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.26 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +0.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.76 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.42 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.39 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.98 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.56 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.21 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.21 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
SYSTIN:                +101.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +92.0°C  (high = +80.0°C, hyst = +75.0°C)  ALARM  sensor = thermistor
AUXTIN0:                +39.0°C    sensor = thermistor
AUXTIN1:                +53.0°C    sensor = thermistor
AUXTIN2:                +95.0°C    sensor = thermistor
AUXTIN3:                +94.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
```

BIOS shows fans (and speeds) without issues
[image]
[image]

hgregorian, 2016-12-20:
Additional insight when poking around in the sys tree:

```
[root@thoth lm-sensors]# find /sys/class/hwmon/hwmon4/device/ -name "pwm*"
/sys/class/hwmon/hwmon4/device/pwm1_auto_point4_temp
/sys/class/hwmon/hwmon4/device/pwm1
/sys/class/hwmon/hwmon4/device/pwm2
/sys/class/hwmon/hwmon4/device/pwm1_auto_point5_temp
/sys/class/hwmon/hwmon4/device/pwm2_step_down_time
/sys/class/hwmon/hwmon4/device/pwm2_floor
/sys/class/hwmon/hwmon4/device/pwm1_weight_temp_sel
/sys/class/hwmon/hwmon4/device/pwm1_target_temp
/sys/class/hwmon/hwmon4/device/pwm2_start
/sys/class/hwmon/hwmon4/device/pwm1_stop_time
/sys/class/hwmon/hwmon4/device/pwm1_step_up_time
/sys/class/hwmon/hwmon4/device/pwm1_enable
/sys/class/hwmon/hwmon4/device/pwm1_weight_temp_step_tol
/sys/class/hwmon/hwmon4/device/pwm2_weight_temp_step
/sys/class/hwmon/hwmon4/device/pwm1_step_down_time
/sys/class/hwmon/hwmon4/device/pwm2_target_temp
/sys/class/hwmon/hwmon4/device/pwm1_auto_point1_pwm
/sys/class/hwmon/hwmon4/device/pwm1_auto_point2_pwm
/sys/class/hwmon/hwmon4/device/pwm1_auto_point3_pwm
/sys/class/hwmon/hwmon4/device/pwm1_auto_point4_pwm
/sys/class/hwmon/hwmon4/device/pwm1_auto_point5_pwm
/sys/class/hwmon/hwmon4/device/pwm2_auto_point1_temp
/sys/class/hwmon/hwmon4/device/pwm2_weight_temp_sel
/sys/class/hwmon/hwmon4/device/pwm2_auto_point2_temp
/sys/class/hwmon/hwmon4/device/pwm2_auto_point3_temp
/sys/class/hwmon/hwmon4/device/pwm2_auto_point4_temp
/sys/class/hwmon/hwmon4/device/pwm2_auto_point5_temp
/sys/class/hwmon/hwmon4/device/pwm1_weight_duty_base
/sys/class/hwmon/hwmon4/device/pwm1_weight_duty_step
/sys/class/hwmon/hwmon4/device/pwm1_weight_temp_step_base
/sys/class/hwmon/hwmon4/device/pwm1_temp_sel
/sys/class/hwmon/hwmon4/device/pwm1_crit_temp_tolerance
/sys/class/hwmon/hwmon4/device/pwm2_weight_temp_step_base
/sys/class/hwmon/hwmon4/device/pwm2_temp_tolerance
/sys/class/hwmon/hwmon4/device/pwm2_weight_temp_step_tol
/sys/class/hwmon/hwmon4/device/pwm2_auto_point1_pwm
/sys/class/hwmon/hwmon4/device/pwm2_auto_point2_pwm
/sys/class/hwmon/hwmon4/device/pwm2_auto_point3_pwm
/sys/class/hwmon/hwmon4/device/pwm2_auto_point4_pwm
/sys/class/hwmon/hwmon4/device/pwm2_auto_point5_pwm
/sys/class/hwmon/hwmon4/device/pwm1_mode
/sys/class/hwmon/hwmon4/device/pwm2_temp_sel
/sys/class/hwmon/hwmon4/device/pwm2_mode
/sys/class/hwmon/hwmon4/device/pwm2_enable
/sys/class/hwmon/hwmon4/device/pwm1_temp_tolerance
/sys/class/hwmon/hwmon4/device/pwm1_weight_temp_step
/sys/class/hwmon/hwmon4/device/pwm2_weight_duty_base
/sys/class/hwmon/hwmon4/device/pwm2_weight_duty_step
/sys/class/hwmon/hwmon4/device/pwm2_stop_time
/sys/class/hwmon/hwmon4/device/pwm2_step_up_time
/sys/class/hwmon/hwmon4/device/pwm1_floor
/sys/class/hwmon/hwmon4/device/pwm1_start
/sys/class/hwmon/hwmon4/device/pwm1_auto_point1_temp
/sys/class/hwmon/hwmon4/device/pwm1_auto_point2_temp
/sys/class/hwmon/hwmon4/device/pwm1_auto_point3_temp
/sys/class/hwmon/hwmon4/device/pwm2_crit_temp_tolerance
```

Shouldn't there be additional pwmN entries to reflect possible fans to monitor/control?

groeck, 2016-12-22:
On Mon, Dec 19, 2016 at 10:28:30PM -0800, Heig Gregorian wrote:
> Additional insight when poking around in the sys tree:
> 
> ```
> Shouldn't there be additional pwmN entries to reflect possible fans to monitor/control?
> 
Only if enabled by the BIOS. The driver detects from chip register which
controls are enabled, and only displays those which are.

Guenter

hgregorian, 2016-12-23:
Thanks for the response.  Would it change things if I told you that when I regular DC fans were plugged in, `sensors-detect` did show them?  Only PWM fans are not detected...

You're probably already aware of this, but Asus implements some form of "Asus Smart Fan Control" (with no apparent way to disable it).  Could this be interfering with the driver's ability to detect PWM fans?  The fan speeds are reported via their BMC (ipmi) protocol.

groeck, 2016-12-23:
On Thu, Dec 22, 2016 at 06:36:31PM -0800, Heig Gregorian wrote:
> Thanks for the response.  Would it change things if I told you that when I regular DC fans were plugged in, `sensors-detect` did show them?  Only PWM fans are not detected...
>
> You're probably [...]
They also have a fan extension board with an nct7802.

Sure, it may well be that they are using the BMC controller for fan control.

Guenter

hgregorian, 2017-01-02:
For fun, used [these](https://www.asus.com/support/faq/1005450/) instructions to disable a couple fan sensors to see if it would produce different results with `sensors-detect` - it did not.  I suppose a next step would be to disable the BMC altogether to see where that gets me, however I do intend to use the BMC, so this is really a non-option.  Looks like I'm stuck letting Asus Smart Fan Control do the work (unless I can find an ipmi command to disable it altogether) or purchase a fan controller.  You wouldn't happen to have any supported recommendations would you?

PeterSurda, 2017-02-14:
I have a similar issue with a similar mainboard (Asus Z10PA-D8). The mainboard has 8 fan headers, 6 of which I have connected, but only 2 show up under /sys. I don't have the BMC board plugged in. The fans speeds show correctly in BIOS (even with the BMC disconnected).

groeck, 2017-02-14:
All information I found about Asus server boards suggest that they are using a BMC for fan control. Asus does not really support Linux, so it is difficult if not impossible to get any information from them. Without such information, it is all but impossible to figure out what is going on, sorry.
Running "sensors-detect" might give an idea if a BMC is present, or maybe some other fan controller (such as nct7802, as mentioned before).

PeterSurda, 2017-02-14:
Well I physically unplugged the BMC board so I'm pretty sure it's not present. I'll try to get the technical documentation from Asus, I did some work on drivers in the past (AMD Mach64 and cx88 infrared remote), maybe they'll give me the specs.

PeterSurda, 2017-02-14:
`modprobe nct7802` does nothing (silently loads, nothing in the kernel log). However paying closer attention to `sensors-detect` brings up this:

> Trying family \`SMSC'...                                     Yes
> Found unknown chip with ID 0x0b00

and

> Found unknown SMBus adapter 8086:8d22 at 0000:00:1f.3.
> Sorry, no supported PCI bus adapters found.

Can I do something about those? Maybe one of them has the fan info.

groeck, 2018-03-11:
The latest Linux kernel supports 8086:8d22. Closing. Please reopen if there is new information.

closed 2018-03-11 by groeck