groeck/nct6775 #49: Add support for NUVOTON NCT6795D

scorpio810, 2017-03-24:
Hi,

I have a brand new MSI AMD B350 motherboard with a Ryzen processor. There is an unknown NCT chip detected.

https://download.tuxfamily.org/qet/forum_img/ryzen1.jpg

Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd352

# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Micro-Star International Co., Ltd MS-7A34 [1.0] (laptop)
# Board: Micro-Star International Co., Ltd B350 TOMAHAWK (MS-7A34)
# Kernel: 4.10.4-vanilla x86_64
# Processor: AMD Ryzen 7 1700X Eight-Core Processor (23/1/1)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd352

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 1022:790b at 0000:00:14.0.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.

[... 366 lines cut ...]
Probing for `Atmel AT30TS00'...                             No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: NVIDIA i2c adapter 0 at 23:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 1 at 23:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 2 at 23:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 6 at 23:00.0 (i2c-7)
Do you want to scan it? (yes/NO/selectively): 

Sorry, no sensors were detected.
This is relatively common on laptops, where thermal management is
handled by ACPI rather than the OS.

scorpio810, 2017-03-25:
I force ID like nct 6793D 
modprobe nct6775 force_id=0xd120

laurent@debian:~$ sudo dmesg | tail 
[14317.528202] nct6775: Found NCT6793D or compatible chip at 0x4e:0xa20

Now I see :

[18:47:33] laurent@debian:~$ sensors
nouveau-pci-2300
Adapter: PCI adapter
GPU core:     +0.90 V  (min =  +0.90 V, max =  +1.11 V)
temp1:        +33.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

nct6793-isa-0a20
Adapter: ISA adapter
in0:                    +0.48 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)                                                                                                                                                                                                             
in11:                   +0.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in12:                   +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in13:                   +0.59 V  (min =  +0.00 V, max =  +0.00 V)  ALARM                                                                                                                                                                                                      
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   847 RPM  (min =    0 RPM)
fan3:                   797 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                   681 RPM  (min =    0 RPM)
SYSTIN:                 +28.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +23.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +30.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +34.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +39.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

groeck, 2017-03-27:
scorpio: The chip is an exclusive for MSI, and Nuvoton tells me they can not give me the datasheet because of an NDA with MSI. They referred me to MSI. Unfortunately, I don't have a contact there, and I don't see how I can contact their technical support without having bought a board. It would be great if you can open a technical support case with MSI and ask for a datasheet.
[We could just assume that the chip is identical to NCT6993D, but I don't really feel comfortable doing that unless MSI doesn't give us a choice and we really have to]

scorpio810, 2017-03-27:
Groeck: I will try to contact MSI support.
On my side how I can help you, isa dump for compare to NCT6793D or other?

groeck, 2017-03-27:
scorpio: You could try running "sudo superiotool -d -e" from coreboot, from the 'devel' branch of my coreboot clone on github.

git clone https://github.com/groeck/coreboot.git
cd coreboot/util/superiotool
git checkout devel
make
sudo ./superiotool -d -e

scorpio810, 2017-03-27:
Groeck: i can try this without any risks?

groeck, 2017-03-27:
Thanks. The chip looks pretty similar to NCT6793D. I am sure there is a difference somewhere, but we won't be able to know what the differences are without datasheet. I'll spend some more time on it in the next couple of days to see if I find problems, otherwise I'll just add the chips as compatible to NCT6793D.

scorpio810, 2017-03-28:
Ticket: NUVOTON NCT6795D datasheet Content: Dear laurentxxxxxxxxxx
Thanks for contacting MSI technical support.

Regarding your concern,sorry to say that we cannot supply it.Or you can
contact the chipset vendor directly.Thanks!

Thanks for your cooperation in advance!

Best Regards,

MSI Technical Support Team

groeck, 2017-03-28:
Nice trick, especially since the chip vendor tells me they can't provide it because of restrictions put upon them by MSI. Guess I'll have to add MSI to my board vendor blacklist.

scorpio810, 2017-03-31:
I saw here cpu-z : https://forum-en.msi.com/index.php?action=dlattach;topic=283169.0;attach=50724
If can help you....
Mainboard Model		B350 TOMAHAWK (MS-7A34) (0x00000573 - 0xD3F0E290)

LPCIO
-------------------------------------------------------------------------

LPCIO Vendor			Nuvoton
LPCIO Model			NCT6795
LPCIO Vendor ID			0x5CA3
LPCIO Chip ID			0xD3
LPCIO Revision ID		0x52
Config Mode I/O address		0x4E
Config Mode LDN			0xB
Config Mode registers	
		00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
	00	FF FF 00 FF FF FF FF 0B FF FF FF FF FF FF FF FF 
	10	FF FF FF FF FF FF FF FF FF FF 30 80 10 00 FF FF 
	20	D3 52 FF 00 00 00 40 13 00 FF 58 01 01 00 00 11 
	30	01 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	40	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	50	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	60	0A 20 0A 40 00 00 FF FF FF FF FF FF FF FF FF FF 
	70	00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
Register space			LPC, base address = 0x0A20

Hardware Monitors
-------------------------------------------------------------------------

Hardware monitor		Nuvoton NCT6795
	Voltage 0		1.00 Volts [0x7D] (CPU VCORE)
	Voltage 1		1.01 Volts [0x7E] (VIN1)
	Voltage 2		3.42 Volts [0xD6] (AVCC)
	Voltage 3		3.38 Volts [0xD3] (3VCC)
	Voltage 4		1.01 Volts [0x7E] (VIN4)
	Voltage 5		0.16 Volts [0x14] (VIN5)
	Voltage 6		0.74 Volts [0x5D] (VIN6)
	Voltage 7		0.00 Volts [0x0] (VIN7)
	Voltage 8		1.00 Volts [0x7D] (VIN8)
	Voltage 9		0.60 Volts [0x4B] (VIN9)
	Voltage 10		1.01 Volts [0x7E] (VIN10)
	Voltage 11		1.01 Volts [0x7E] (VIN11)
	Voltage 12		0.74 Volts [0x5D] (VIN12)
	Voltage 13		0.90 Volts [0x71] (VIN13)
	Voltage 14		0.91 Volts [0x72] (VIN14)
	Temperature 0		36 degC (96 degF) [0x24] (SYSTIN)
	Temperature 1		40 degC (104 degF) [0x50] (CPUTIN)
	Temperature 3		40 degC (104 degF) [0x28] (TMPIN3)
	Temperature 5		37 degC (98 degF) [0x25] (TMPIN5)
	Temperature 6		36 degC (96 degF) [0x24] (TMPIN6)
	Fan 0			4272 RPM [0x10B0] (SYSFANIN)
	Fan 1			860 RPM [0x35C] (CPUFANIN)
	Fan 3			1160 RPM [0x488] (AUXFANIN1)
	Fan 4			1425 RPM [0x591] (AUXFANIN2)
	Fan 5			607 RPM [0x25F] (AUXFANIN3)
Hardware registers	
Register space			LPC, base address = 0x0A20
bank 0	
		00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
	00	03 64 03 64 00 FF FF FF FF FF FF FF FF FF FF FF 
	10	03 93 03 93 03 93 03 93 40 E4 00 72 72 4B C0 00 
	20	7D 7E D6 D3 7F 14 5D 24 FF FF FF DA 00 00 00 00 
	30	00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF 
	40	01 3F 07 FF DF 00 3F FF 2D FF FF 10 00 55 00 A3 
	50	05 FF FF FF FF FF FF FF C1 FF FF FF FF 02 02 00 
	60	54 F0 FF FF FF 01 00 FF FF FF FF FF FF FF FF FF 
	70	00 0A 00 36 80 36 80 36 80 36 80 36 80 36 80 FF 
	80	FF 7F 3F BF FF 05 05 05 00 00 00 00 00 00 00 00 
	90	0A 00 00 00 00 FD FD FD 0A 95 EF 80 FF 40 46 C4 
	A0	0E FF 00 00 FF 00 00 FF 00 00 80 66 06 06 00 00 
	B0	00 00 00 00 00 00 2F 00 04 00 00 0A B4 80 04 00 
	C0	FF FF FF FF FF FF FF FF FF 06 02 05 FF FF FF FF 
	D0	05 05 05 05 00 00 00 00 00 00 00 00 FD FD FD FD 
	E0	55 80 00 00 00 FF FF FF FF FF 00 00 00 00 00 00 
	F0	00 00 00 00 00 00 00 FF 1C 00 4C 03 9F FF FF FF 
bank 1	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
bank 2	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
bank 3	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
bank 4	
	80	7D 7E D6 D2 7E 14 5D D5 00 E4 00 72 72 4B C0 00 
	90	24 25 28 80 1F FE 00 2F 80 36 36 36 36 36 1F 36 
bank 5	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
bank 6	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
bank 7	
	80	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
	90	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

scorpio810, 2017-04-03:
[NCT6793_NCT6795.txt](https://github.com/groeck/nct6775/files/890653/NCT6793_NCT6795.txt)

OursDesCavernes, 2017-04-05:
Hello,
I have a similar chip in a Gigabyte B250M PRO-VH
I found this thread because I also have:
```
Trying family `VIA/Winbond/Nuvoton/Fintek'... Yes
Found unknown chip with ID 0xd352
```
`modprobe nct6775 force_id=0xd120` works:
```
#> sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +35.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +35.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +35.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +36.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

nct6793-isa-0a20
Adapter: ISA adapter
in0:                    +0.66 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.13 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1038 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +111.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                +127.5°C  (high = +80.0°C, hyst = +75.0°C)  ALARM  sensor = CPU diode
AUXTIN0:                +42.0°C    sensor = CPU diode
AUXTIN1:               +114.0°C    sensor = thermistor
AUXTIN2:               +115.0°C    sensor = thermistor
AUXTIN3:                 -3.0°C    sensor = thermistor
PECI Agent 0:           +36.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled

amdgpu-pci-0100
Adapter: PCI adapter
fan1:         687 RPM
temp1:        +42.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

```
Superiotool output:
`Found Nuvoton NCT6795D-M (id=0xd352) at 0x4e`
[superiotool.txt](https://github.com/groeck/nct6775/files/900990/superiotool.txt)

groeck, 2017-04-06:
OursDesCavernes: I am quite sure that this is an MSI board, not Gigabyte. Gigabyte does not appear to sell any "PRO-VH" boards, and Nuvoton says that the chip was explicitly produced for MSI.
Temperatures seem to be a bit on the high side ;-). This is not much of concern, though - I already know that the temperature labels have changed. PECI Agent 0 is the "real" CPU temperature, most of the others may be disconnected on this board.

scorpio810, 2017-04-06:
Set in manual speed fan in the BIOS 
nouveau-pci-2300
Adapter: PCI adapter
GPU core:     +0.90 V  (min =  +0.90 V, max =  +1.11 V)
temp1:        +33.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

nct6793-isa-0a20
Adapter: ISA adapter
in0:                    +0.84 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.72 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.60 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   870 RPM  (min =    0 RPM)
fan3:                  1208 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                   573 RPM  (min =    0 RPM)
SYSTIN:                 +29.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +30.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +42.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +34.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +44.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

[image]

nagisa, 2017-04-08:
The same chip is on the MSI B350M Mortar Arctic. Nagged a bit to MSI support for extra pressure.

The CPU temps and fan speeds seem to reflect reality.

nagisa, 2017-04-08:
Checked fan speeds against BIOS. they are pretty much spot on (within 100 RPM of the RPM displayed by the BIOS). Mobo temperature matches and CPU temperature seems to match as well, though can’t tell for sure because CPU seems to get some workload while in BIOS (and temp rises).

padrino121, 2017-04-09:
At this point the only issue I have found is the last system fan does not seem to be exposed, pwm5 is the 3rd system fan, I cannot get to the 4th.

groeck, 2017-04-09:
padrino121: Unfortunately that is one of those problems which would likely require a datasheet to fix.

cpupirate, 2017-04-11:
searching for a solution to get conky working on my new build, thought I would add some info to this, I seem to have the same chipset on MSI Pro Carbon x370

# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Micro-Star International Co., Ltd MS-7A32 [1.0] (laptop)
# Board: Micro-Star International Co., Ltd X370 GAMING PRO CARBON (MS-7A32)
# Kernel: 4.10.1-041001-generic x86_64
# Processor: AMD Ryzen 7 1700X Eight-Core Processor (23/1/1)
..
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd352

So is the command referenced above likely to be the longer term solution?
"modprobe nct6775 force_id=0xd120"

groeck, 2017-04-11:
@cpupirate: If you are not happy that temperature sensor labels are not displayed for NCT6795D, yes. Please note, though, that the labels displayed with force_id=0xd120 are at least partially wrong, and you still won't see pwm4.

scorpio810, 2017-04-24:
FYI, "in0" entry matches CPU core voltage.

mglinux, 2017-05-15:
This is my output. It´s an ASRock fatal1ty gaming K4 (B350 chipset)

nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.36 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.29 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.31 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +1.10 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.47 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.30 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.94 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +1.07 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   817 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +33.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +34.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                 +7.5°C    sensor = thermistor
AUXTIN1:                +29.0°C    sensor = thermistor
AUXTIN2:                +22.0°C    sensor = thermistor
AUXTIN3:                -29.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

Systin and cputin seems correct. But none of the values represents my real vcore. It should be 1.24v on load

scorpio810, 2017-05-15:
Isn't the same nuvoton I/O chip : nct 6779
http://www.tweaktown.com/reviews/8173/asrock-fatal1ty-ab350-gaming-k4-motherboard-review/index4.html

mglinux, 2017-05-16:
@scorpio810 

You are right. Confusing Sensors... But the Problem itself seems similar

scorpio810, 2017-05-16:
Your chip was added 3 year ago : 
http://lm-sensors.lm-sensors.narkive.com/cGu8iaj5/patch-sensors-conf-default-add-default-voltage-settings-for-nct6791

https://www.spinics.net/lists/lm-sensors/msg37631.html

groeck, 2017-05-16:
@mglinux: You mean you don't see pwm4 ? The voltages need to be scaled in software, as with all other sensors.

mglinux, 2017-05-16:
@groeck 

Cputemp is fine. Systin sometimes freezes or shows negative values. I can See fan speeds and 3.3 and 12v too. 

The only thing I cant See is the real vcore of the CPU. Vcore value shows +0.5 e.g. 

And none oft the other values lije in1,2... is the vcore.

groeck, 2017-05-16:
pwm and fan detection for NCT6793D and NCT6795D has been improved. Please test and report.
pwm entries are not shown in sensors, only as sysfs attributes.
No idea yet what might cause the SYSTIN problem; that seems to be common across several boards and suggests a generic problem (bug ?) in the driver; see 'very cold mainboard' issue. Please report and track there.

groeck, 2017-05-19:
Should now work. Tested on MSI B350 Tomahawk.

closed 2017-05-19 by groeck

scorpio810, 2017-05-20:
With latest commit : 
nct6795-isa-0a20
Adapter: ISA adapter
in0:                    +1.24 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.80 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +1.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.60 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   857 RPM  (min =    0 RPM)
fan3:                   878 RPM  (min =    0 RPM)
fan4:                   976 RPM  (min =    0 RPM)
fan5:                   736 RPM  (min =    0 RPM)
CPUTIN:                 +27.0°C  (high = +127.0°C, hyst = +105.0°C)  sensor = thermistor
AUXTIN0:                +37.0°C  (high = +127.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +20.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +44.5°C  (high = +127.0°C, hyst = +105.0°C)
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled                                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                              
nouveau-pci-2300                                                                                                                                                                                                                                                              
Adapter: PCI adapter                                                                                                                                                                                                                                                          
GPU core:     +0.90 V  (min =  +0.90 V, max =  +1.11 V)                                                                                                                                                                                                                       
temp1:        +30.0°C  (high = +95.0°C, hyst =  +3.0°C)                                                                                                                                                                                                                       
                       (crit = +105.0°C, hyst =  +5.0°C)                                                                                                                                                                                                                      
                       (emerg = +135.0°C, hyst =  +5.0°C)

;-)

groeck, 2017-05-20:
Added another small fix to display SYSTIN as well. CPU temperature on MSI boards is reported with SMBUSMASTER 0. Depending on the BIOS version and the CPU type, the reported temperature may be 20 degrees C too high (this is a known problem with 1700X and 1800X CPUs).
AUXTIN1..3 input pins are also used as VIN5..7 voltage input pins. It is unknown which one is active on a given board.

scorpio810, 2017-05-25:
$ sensors
nct6795-isa-0a20
Adapter: ISA adapter
in0:                    +1.24 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +1.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.60 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   895 RPM  (min =    0 RPM)
fan3:                   973 RPM  (min =    0 RPM)
fan4:                  1105 RPM  (min =    0 RPM)
fan5:                   840 RPM  (min =    0 RPM)
SYSTIN:                 +34.0°C  (high = +127.0°C, hyst =  +0.0°C)  sensor = CPU diode
CPUTIN:                 +33.0°C  (high = +127.0°C, hyst = +105.0°C)  sensor = thermistor
AUXTIN0:                +42.0°C  (high = +127.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +20.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +51.5°C  (high = +127.0°C, hyst = +105.0°C)
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM                                                                                                                                                                                                                                                  
beep_enable:           disabled                                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                              
nouveau-pci-2300                                                                                                                                                                                                                                                              
Adapter: PCI adapter                                                                                                                                                                                                                                                          
GPU core:     +0.90 V  (min =  +0.90 V, max =  +1.11 V)                                                                                                                                                                                                                       
temp1:        +35.0°C  (high = +95.0°C, hyst =  +3.0°C)                                                                                                                                                                                                                       
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

;-)

groeck, 2017-05-25:
CPU temperature is "SMBUSMASTER 0". The reported temperature may be 20 degrees C too high; at least it is on the two MSI boards I have myself (B350 Tomahawk and B350M-DHC) with 1700X. I am hesitant to recommend configuring a temperature offset; automatic fan control is configured using the higher temperatures. Changing the temperature offset would mess that up.

eeyrjmr, 2017-06-04:
On an X370 GAMING PRO CARBON with a Ryzen5 1600:

```
modprobe nct6775 force_id=0xd120
fluidmotion openbox # sensors
nct6793-isa-0a20
Adapter: ISA adapter
in0:                    +1.45 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.79 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                   796 RPM  (min =    0 RPM)
fan3:                  1862 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                  1939 RPM  (min =    0 RPM)
SYSTIN:                 +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +37.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +37.5°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +56.0°C    sensor = thermistor
AUXTIN3:                 -2.0°C    sensor = thermistor
SMBUSMASTER 0:          +36.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

```
The odd thing for me is, SMBUSMASTER reads +36, the same as CPUTIN.  AUXTIN2 reads 56C 

The "higher temperature" (whichever sensor it is...) isn't a bug, it was an intended offset by AMD of the X series CPU, the ones with higher thermal wattage:

https://community.amd.com/community/gaming/blog/2017/03/13/amd-ryzen-community-update

> Specifically, the AMD Ryzen™ 7 1700X and 1800X carry a +20°C offset between the tCTL° (reported) temperature and the actual Tj° temperature. In the short term, users of the AMD Ryzen™ 1700X and 1800X can simply subtract 20°C to determine the true junction temperature of their processor. No arithmetic is required for the Ryzen 7 1700. Long term, we expect temperature monitoring software to better understand our tCTL offsets to report the junction temperature automatically.

scorpio810, 2017-06-04:
> The odd thing for me is, SMBUSMASTER reads +36, the same as CPUTIN. AUXTIN2 reads 56C
Not the same values here with Tomahawk and 1700X CPU.
```
SYSTIN:                 +30.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +30.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +37.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +20.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +47.0°C  
```

To be really sure, someone can try to control actual CPU temp and Mobo probes on Windows with HWinfo or other software and compare on Gnu/Linux with sensor program?
Sorry I can't.  :D

groeck, 2017-06-05:
My boards: B350 TOMAHAWK (MS-7A34) and B350M MORTAR (MS-7A37), both with 1700X. in0 was set by the BIOS on its own with some overclocking attempts (which I since have given up until things stabilize on the BIOS side).

CPUTIN and AUXTIN0 move up and down depending on system load, but do not reflect the CPU temperature. I suspect AUXTIN0 may be a voltage regulator. CPUTIN might be a discrete temperature sensor located close to the CPU, and CPUTIN might be a discrete temperature sensor somewhere on the board. I suspect that AUXTIN[-3] are connected to voltages, not to temp sensors. I still have not been able to figure out how to determine if the input pins are used as temp sensors or as voltage sensors.

Note that one can not determine from Super-IO temperature values if a sensor is connected to the CPU or not. One really needs k10temp working for Ryzen. Unfortunately, the information how to do that is still only available under NDA.

peterbecich, 2017-06-08:
@mglinux : maybe this is similar to your situation.
Mobo is ASRock AB350m.  CPU is Rysen 1700.  Kernel is 4.9.

```
$ sudo sensors-detect
# sensors-detect revision 6209 (2014-01-14 22:51:58 +0100)
# Board: ASRock AB350M Pro4
 
This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.
 
Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No
 
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
```
A Nuvoton is found:
```
Found `Nuvoton NCT5532D/NCT6779D Super IO Sensors'          Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

```
Full output of `sensors-detect`: https://pastebin.com/gSCHQCtk

Unfortunately I only get CPUTIN and the like, not the core temps: https://pastebin.com/JRV2xBrc

```
$ sensors
nct6779-isa-0290
Adapter: ISA adapter
....
SYSTIN:                 +26.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +28.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +82.0°C    sensor = thermistor
AUXTIN1:                +99.0°C    sensor = thermistor
AUXTIN2:                +24.0°C    sensor = thermistor
AUXTIN3:                -23.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled
```

groeck, 2017-06-08:
Reporting core temperatures for Ryzen chips is a matter of adding AMD Family 17h support to k10temp. As mentioned before, the necessary information to do that is only available under NDA. I'll be happy to accept a patch for the k10temp driver, or to make the necessary changes myself if/when the information is released.

peterbecich, 2017-06-08:
Thank you @groeck .  Sorry I overlooked your explanation of k10temp.

groeck, 2017-06-09:
No worries. It _is_ highly annoying that AMD does not release those specifications :-(.

rigred, 2017-06-25:
Has anyone yet figured out the correct offsets for the for the nuvoton line of SIO's on ryzen platforms?
This is only slightly relevant to this thread.

I have an nct6779 and if anyone knows of someone that already got the compute parameters figured out let me know.

Else I will go off by myself and figure it out for the board I have.

groeck, 2017-06-25:
I don't even know what you mean with "nuvoton line of SIO's" :-(.

rigred, 2017-06-25:
Well specifically the Nuvoton nct6779 Super IO

Here's my output right now. I've started to create a custom sensor config, but figuring out the exact voltage input inorder to calculate the offsets for the 5V/12V and temperatures etc is tricky.
I do have a datasheet for the device, but obviously the voltage and temp offsets are actually calculated in the bios.
`nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.44 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.36 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.36 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.46 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.25 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.81 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +1.09 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.93 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
CPU Fan (Aux):            0 RPM  (min =    0 RPM)
CPU Fan:                997 RPM  (min =  950 RPM)
Case Fan 1:            1103 RPM  (min =    0 RPM)
Case Fan 2:            1148 RPM  (min =    0 RPM)
HDD Fan:                844 RPM  (min =    0 RPM)
SYSTIN:                 +28.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPU:                    +30.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +15.0°C    sensor = thermistor
VRM:                    +37.0°C    sensor = thermistor
AUXTIN2:                +21.0°C    sensor = thermistor
AUXTIN3:                -26.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
`

groeck, 2017-06-25:
Assuming you mean scaling factors, the best approach is still to monitor voltages in the BIOS for a while and observe voltage differences. The voltage "jumps" are multiples of the scaling factors, which you then can use to map inX with the respective voltages and to calculate the scaling factors.

rigred, 2017-06-25:
Thanks that's what I will be doing then. 
Probably also going to be referring to windows hwinfo and those tools and try some RW everything trickery.
Hopefully I can provide a config file for the board (ASRock X370 Gaming K4) once done. 

Is there anywhere specific where I should share my config files for others to find them easily?

groeck, 2017-06-25:
Best (only) means right now is to send an e-mail to linux-hwmon@vger.kernel.org.

rigred, 2017-06-27:
Well just for the sake of it here is the config for the ASRock X370 Gaming K4.
These may apply to other nct6779D boards but so far this is purely specific to this board

[lm_sensors config for ASRock X370 Gaming K4 board](https://gist.github.com/rigred/000cb8486dfae1972154ec9bf8355ac0)

mglinux, 2017-07-07:
@rigred 

Holy sh... Your configs works for my Asrock B350 gaming K4.

Thank you!!!

_Gesendet von meinem Samsung SM-G900F mit [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

tacchinotacchi, 2017-08-20:
Tried to keep pressuring MSI, that's as much as I can do now

jbtrystram, 2017-08-29:
Hi,

Same chip ID detected (0xd352) but on an kaby lake system. Mobo is MSI B150M PRO VDH. Is there a way I can help ?

groeck, 2017-08-29:
Not sure. What doesn't work when you run the driver here or the latest upstream kernel (v4.13-rc7) ?

EUA, 2017-11-11:
Here is Gigabyte AB350 Gaming ITX/AC 
sensors-detect can't find the SuperIO. After modprobing nct6775

```
[root@Ryzen death]# sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +51.0°C  

nct6793-isa-0290
Adapter: ISA adapter
in0:                    +0.58 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in2:                    +3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in5:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                    +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.46 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.13 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  2542 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                     0 RPM
SYSTIN:                +116.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
CPUTIN:                 +51.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +45.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               +108.0°C    sensor = thermistor
AUXTIN2:               +106.0°C    sensor = thermistor
AUXTIN3:               -128.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled

radeon-pci-0b00
Adapter: PCI adapter
temp1:        +57.5°C  (crit = +120.0°C, hyst = +90.0°C)

```

SVolkmann, 2018-01-16:
on my board seems to work corect:
stephan@stephan-MS-7A75:~$ sensors
nct6793-isa-0a20
Adapter: ISA adapter
in0:                    +1.22 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.13 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +1.07 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.61 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1077 RPM  (min =    0 RPM)
fan3:                   696 RPM  (min =    0 RPM)
fan4:                   557 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +27.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +48.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +19.0°C    sensor = thermistor
AUXTIN1:                +22.0°C    sensor = thermistor
AUXTIN2:               +127.0°C    sensor = thermistor
AUXTIN3:                 -3.0°C    sensor = thermistor
PECI Agent 0:           +47.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +60.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +59.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +60.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +50.0°C  (high = +80.0°C, crit = +100.0°C)
based on inxi, also all fine:
stephan@stephan-MS-7A75:~$ inxi -F
System:    Host: stephan-MS-7A75 Kernel: 4.14.0-16-generic x86_64 bits: 64 Desktop: Xfce 4.12.3
           Distro: Ubuntu Bionic Beaver (development branch)
Machine:   Device: desktop Mobo: MSI model: Z270 GAMING PLUS (MS-7A75) v: 1.0 serial: N/A
           UEFI: American Megatrends v: 1.40 date: 11/17/2017
CPU:       Quad core Intel Core i7-7700K (-MCP-) cache: 8192 KB
           clock speeds: max: 4500 MHz 1: 4200 MHz 2: 4200 MHz 3: 4200 MHz 4: 4200 MHz
Graphics:  Card: Intel HD Graphics 630
           Display Server: x11 (X.Org 1.19.5 ) drivers: modesetting (unloaded: fbdev,vesa)
           Resolution: 1920x1200@59.95hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics 630 (Kaby Lake GT2) version: 4.5 Mesa 17.3.1
Audio:     Card Intel 200 Series PCH HD Audio driver: snd_hda_intel Sound: ALSA v: k4.14.0-16-generic
Network:   Card: Intel Ethernet Connection (2) I219-V driver: e1000e
           IF: enp0s31f6 state: up speed: 1000 Mbps duplex: full mac: 4c:cc:6a:f4:92:2e
Drives:    HDD Total Size: 250.1GB (4.5% used)
           ID-1: /dev/nvme0n1 model: Samsung_SSD_960_EVO_250GB size: 250.1GB
Partition: ID-1: / size: 228G used: 11G (5%) fs: ext4 dev: /dev/nvme0n1p2
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 49.0C mobo: 27.0C
           Fan Speeds (in rpm): cpu: 1075 fan-1: 0 fan-3: 706 fan-4: 572 fan-5: 0
Info:      Processes: 179 Uptime: 26 min Memory: 567.4/6923.7MB Client: Shell (bash) inxi: 2.3.45 

cu, stephan

DocMAX, 2018-03-26:
Is anybody running PWM in DC mode (3pin)?

samiraguiar, 2018-05-25:
My two cents on this one. I've got a B350 Tomahawk and Ryzen 7 2700x.

I ran the command mentioned above (if I understood correctly, that's currently the only way since MSI is not helping):
`sudo modprobe nct6775 force_id=0xd120`

And sensors:
```
nct6795-isa-0a20
Adapter: ISA adapter
in0:                    +0.77 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.44 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.14 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.80 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.59 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.51 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                   289 RPM  (min =    0 RPM)
fan2:                     0 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +31.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +33.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +37.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +35.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
SMBUSMASTER 0:          +41.0°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

groeck, 2018-05-25:
Support for NCT6795D has been added to the upstream kernel in May 2017, with commit 419220dc48699. It is also supported by the driver here. It runs just fine on my B350 Tomahawk with Ryzen 1700X and v4.14.37, and on B350M Mortar with Ryzen 2700X and v4.14.38. The output above suggests that NCT6795 is supported by the kernel installed in your system. It is not clear to me why you would need force_id. Have you tried without ?

samiraguiar, 2018-05-26:
@groeck Do you mean it should work by just running `sudo modprobe nct6775` without the `force_id` or that `sensors-detect` should have detected it?

groeck, 2018-05-26:
1) yes, as you confirmed.
2) depends on the version of sensors-detect

samiraguiar, 2018-05-26:
Ok, thank you for the help!