groeck/nct6775 #50: NCT6791D sometimes reports 0 readings

mooop12, 2017-03-25:
When using `watch -n 1 sensors` I can see that every few seconds all values reported are 0 like so:

```
nct6791-isa-0290
Adapter: ISA adapter
Vcore:                  +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in1:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
AVCC:                   +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
+3.3V:                  +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
in4:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in5:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in6:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
3VSB:                   +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
Vbat:                   +4.08 V  (min =  +4.08 V, max =  +4.08 V)  ALARM
in9:                    +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in10:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in11:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in12:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in13:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
in14:                   +2.04 V  (min =  +2.04 V, max =  +2.04 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)  ALARM
fan2:                     0 RPM  (min =    0 RPM)  ALARM
fan3:                     0 RPM  (min =    0 RPM)  ALARM
fan4:                     0 RPM  (min =    0 RPM)  ALARM
fan5:                     0 RPM  (min =    0 RPM)  ALARM
fan6:                     0 RPM  ALARM
SYSTIN:                  -1.0°C  (high =  -1.0°C, hyst =  -1.0°C)  sensor = thermistor
CPUTIN:                  +0.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
AUXTIN0:                 +0.0°C    sensor = thermistor
AUXTIN1:                 -1.0°C    sensor = thermistor
AUXTIN2:                 -1.0°C    sensor = thermistor
AUXTIN3:                 -1.0°C    sensor = thermistor
PECI Agent 0:            +0.0°C  
PCH_CHIP_CPU_MAX_TEMP:   -1.0°C  
PCH_CHIP_TEMP:           -1.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           enabled
```

What might be causing this?

groeck, 2017-03-25:
Interesting. I don't think I have ever seen this before. What is the board type ?
Is this an ASUS board, by any chance, and you boot with acpi_enforce_resources=lax ?
If you run "sudo watch -n 1 isadump -y 0x295 0x296", do you see something similar ?

mooop12, 2017-03-25:
Yes, the board is an ASUS one, specifically Maximus VII Ranger. I had to use `acpi_enforce_resources=lax`, otherwise the readings would not show up at all (`sensors-detect` detected the nct6791d, but `sensors` didn't list it). The array outputted by `isadump` also changes every few seconds to being filled only with `ff`.

groeck, 2017-03-25:
What is most likely happening is that the BIOS accesses the chip continuously and somehow blocks Linux from accessing it. Unfortunately, I don't think there is anything we can do about that. You could try to contact ASUS and ask them if there is a remedy, but ASUS specifically does not support Linux on their consumer boards, so it is unlikely that they are going to be helpful.

groeck, 2017-03-26:
Can you send me the DSDT for your board ? That should work with
    sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
    iasl -d dsdt.dat
The output is in dsdt.dsl.

No promises, but maybe there is a means to tell the BIOS on ASUS boards to leave the chip alone while the kernel accesses it. Maybe I can figure out what it is by collecting the DSDT from various ASUS boards.

mooop12, 2017-03-26:
Here's the raw dsdt.dat: [dsdt.zip](https://github.com/groeck/nct6775/files/870885/dsdt.zip). iasl is giving me segfaults, can you parse it on your side?

Also, I'm not sure if it's related in any way, but when booting the system I get this message:

```
[    1.492274] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20160930/psargs-359)
[    1.492297] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT0.SPT5._GTF] (Node ffff88040e0e9578), AE_NOT_FOUND (20160930/psparse-543)
[    1.492852] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20160930/psargs-359)
[    1.492874] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT0.SPT5._GTF] (Node ffff88040e0e9578), AE_NOT_FOUND (20160930/psparse-543)
```

People mentioned that updating the BIOS might fix this, but I'm already running the latest BIOS.

groeck, 2017-03-26:
I was able to disassemble it. As far as I can see, the BIOS locks IO register access to the NCT6791 on a regular basis, whenever it executes method _E12 (this is a method under scope _GPE; I read it as "General Purpose Event 12", but that may be wrong, and I don't really know what it means). While we could add some code to wait for this lock to be released, this would be a one-way route: There is no guarantee that the BIOS does not set the lock after the kernel driver checked it. On top of that, the BIOS also changes the selected register bank.
A cleaner fix would be to acquire the ACPI mutex used by the BIOS. I don't know how to do that, or if it is even possible.
Either case, for the time being, unfortunately we'll have to assume that this board simply does not work with the kernel driver.

mooop12, 2017-03-27:
That's a shame. There is some fan control available in the BIOS, but it's nowhere near the capabilities of manual fine-tuning. I guess I'll have to live with it, at least until I upgrade my system. Thank you for your help though.

closed 2017-03-27 by mooop12