groeck/nct6775 #51: very cold mainboard

trezqers, 2017-04-14:
hi

Sometimes I get MB temp read as -60. I'm using MSI B85-G43 mainboard with NCT6779D on it.

It is temp1 sensor, normal values matches board temp shown in BIOS and it is only one *TIN connected.

After boot read is ok and after some time (stations are working 24/7) sometimes it starts to get -60. It happens rarely but this year did on 3 systems and I did not noticed any clue why, there are no changes in any other sensors values nor any other aspect of system.

When it starts that it does not change back, but after reload of module (rmmod & modprobe) proper values get back

I noticed also, that as changing temp1_offset changes value of temp1_input, when it starts with -60, changing offset does not change input any more

do you have any idea why or whats happing?

groeck, 2017-04-14:
Interesting problem. Can you send output from "superiotool -d -e" from before and after the problem is seen ? Also, can you send me your DSDT ?
One possibility is that the temperature input source is changed. This would be bank 6, registers 0x21..0x26. No idea though why that would happen.

trezqers, 2017-04-18:
I'm sorry but I had to reload module on friday, it’s a production system, and could not left it for weekend, so now I must wait for another event to dump wrong state :(

but I tried superiotool to be prepared and it said: No Super I/O found, it is NCT6779D, I checked a label on chip, and with -V there is something like this:

Probing for Nuvoton Super I/O at 0x4e...
  Failed. Returned data: chip_id=0xc563

I changed nuvoton.c:397
`{0xc562, "NCT6779D", {`
to
`{0xc563, "NCT6779D", {`

and now I can dump something I hope real :)
[dump](http://trez.name/superio/superio.dump.txt)

When another event occurs I will update with superio dump from it

If with DSDT you mean /sys/firmware/acpi/tables/DSDT, you can check it here:
[dsdt.bin](http://trez.name/superio/dsdt.bin)
[with iasl decoded dsdt.dsl](http://trez.name/superio/dsdt.dsl)

trezqers, 2017-04-19:
ok, I have another one, this time temp is moving from -60 to -40 and sometimes for a second it jumps to for exmaple +113

[superio dump](http://trez.name/superio/superio.wrong_dump.txt) with -60 in sensors

I have also noticed, that it starts to show the same value as AUXTIN3:
sensors -c /dev/null
nct6779-isa-0a00
Adapter: ISA adapter
in0:                    +0.87 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.90 V, max =  +1.10 V)
in2:                    +3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.05 V  (min =  +0.98 V, max =  +1.20 V)
in5:                    +0.90 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.02 V  (min =  +0.80 V, max =  +1.60 V)
in10:                   +0.90 V  (min =  +0.50 V, max =  +1.50 V)
in11:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in12:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.74 V  (min =  +0.67 V, max =  +0.82 V)
in14:                   +0.78 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  2000 RPM  (min =  700 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                  1450 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 -60.0 C  (high = +50.0 C, hyst = +45.0 C)  sensor = CPU diode
CPUTIN:                +127.5 C  (high = +80.0 C, hyst = +75.0 C)  ALARM  sensor = CPU diode
AUXTIN0:                +22.0 C    sensor = thermistor
AUXTIN1:                +25.0 C    sensor = thermistor
AUXTIN2:               +127.0 C    sensor = thermistor
AUXTIN3:                -60.0 C    sensor = thermal diode
PECI Agent 0:           +41.5 C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0 C  
PCH_CHIP_TEMP:           +0.0 C  
PCH_CPU_TEMP:            +0.0 C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

I checked it for some time and both get the same value all the time, normally auxtin3 shows stupid things, mostly on minus, but I did not bother, assumed it is not connected to anything:
nct6779-isa-0a00
Adapter: ISA adapter
in0:                    +0.87 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.90 V, max =  +1.10 V)
in2:                    +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.31 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.06 V  (min =  +0.98 V, max =  +1.20 V)
in5:                    +0.88 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.44 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.02 V  (min =  +0.80 V, max =  +1.60 V)
in10:                   +1.05 V  (min =  +0.50 V, max =  +1.50 V)
in11:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in12:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.74 V  (min =  +0.67 V, max =  +0.82 V)
in14:                   +0.78 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  2042 RPM  (min =  700 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                  1560 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +27.0 C  (high = +50.0 C, hyst = +45.0 C)  sensor = CPU diode
CPUTIN:                +127.5 C  (high = +80.0 C, hyst = +75.0 C)  ALARM  sensor = CPU diode
AUXTIN0:                +22.0 C    sensor = thermistor
AUXTIN1:                +38.0 C    sensor = thermistor
AUXTIN2:               -128.0 C    sensor = thermistor
AUXTIN3:                -45.0 C    sensor = thermal diode
PECI Agent 0:           +29.0 C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0 C  
PCH_CHIP_TEMP:           +0.0 C  
PCH_CPU_TEMP:            +0.0 C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

groeck, 2017-04-26:
> ok, I have another one, this time temp is moving from -60 to -40 and sometimes for a second it jumps to for exmaple +113
> 
> [superio dump](http://trez.name/superio/superio.wrong_dump.txt) with -60 in sensors

Just to let you know that I have not forgotten you.

The SYSTIN register (bank 0 register 0x27) shows 0xc4 in both cases, which
matches the -60 degrees C. In the good case, it looks like it takes the
temperature value from the alternate SYSTIN register at bank 4, 0x90.
AUXTIN3 would be at page 4, register 0x95, and shows 0xc4 (-60 degrees C)
in both cases.

The temperature source for SYSTIN (page 6 register 0x21) is listed as 0x6
in both cases, which would make it AUXTIN3 (what a coincidence). Ultimately
that means that both AUXTIN3 and SYSTIN should always display the same
temperature (which would be around -60 degrees C).

All this makes makes me wonder if there is a software bug lurking somewhere,
but I have not found it yet.

Guenter

mglinux, 2017-05-21:
It´s rather strange. Now the CPUTIN shows negative values too. This wasn´t before only systin :-)

groeck, 2017-05-21:
The latest version of the driver creates a debugfs file in /sys/kernel/debug/hwmonX/registers, where X is the hwmon device index. It would be great if you can provide the contents of this file in the ok and error cases.

groeck, 2018-03-11:
No further information. Closing.

closed 2018-03-11 by groeck