groeck/nct6775 #53: MSI x370 gaming pro carbon not known

hschraep, 2017-08-04:
Hi looks like the chip on this motherboard was not known. Get this output during sensors-detect:

Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd352

groeck, 2017-08-04:
Did you try this driver ? The message is from sensors-detect, which doesn't mean anything in the context of the driver.

hschraep, 2017-08-05:
OK, yes i have installed this driver and get this output if i type sensors:

```
nct6795-isa-0a20
Adapter: ISA adapter
in0:                    +0.50 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.80 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.23 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.62 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.68 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                  1970 RPM  (min =    0 RPM)
fan2:                   704 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +33.0°C  (high = +127.0°C, hyst =  +0.0°C)  sensor = CPU diode
CPUTIN:                 +30.0°C  (high = +127.0°C, hyst = +105.0°C)  sensor = thermistor
AUXTIN0:                +37.0°C  (high = +127.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +48.0°C    sensor = thermistor
AUXTIN3:                 -2.0°C    sensor = thermistor
SMBUSMASTER 0:          +27.0°C  (high = +127.0°C, hyst = +105.0°C)
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

But i think the fans are not correct:

fan1:                  1970 RPM  (min =    0 RPM)
fan2:                   704 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)

So fan1 and fan2  are interchanged.  Within the bios/uefi fan2 (pump) runns with 1970rpm and fan1 (CPU)with 704.
The next thing is that fan5 displays nothing but there is also a fan connected and runs also with 700rpm

So thats are why i think the driver are not correct recognised.

hschraep, 2017-08-05:
Within dmesg i found this:
[    6.244110] nct6775: Found NCT6795D or compatible chip at 0x4e:0xa2

groeck, 2017-08-05:
Regarding fan1 / fan2 "interchanged": Per Nuvoton datasheets, sysfan is the first fan, cpufan is the second fan. This has been like that from the first chip in the series (NCT6775). It may well be that MSI decided to reverse the order in the BIOS, but that does not mean that anything is wrong with the driver. Similar, "fan5" refers to the 5th fan supported by the chip (aka AUXFAN2 in Nuvoton's datasheets). This is not necessarily associated with the "fan5" header on a given board. The board vendor decides how to connect chip outputs with fan headers on the board, with no regard for chip numbering. If you dislike the default fan names, you can configure any name you want in /etc/sensors3.conf.

As for the conclusion that the chip is not supported by the driver, I can not follow your logic. The driver reports that it identified the chip, the driver instantiates without error and reports data. Yet, you conclude that the driver doesn't support the chip. Please elaborate and explain how you reached that conclusion.

I did find that fan6 support detection is wrong. I am working on a fix.

groeck, 2017-08-05:
Please test the latest driver version and let me know if fan6 now works.

hschraep, 2017-08-05:
Hi groeck,
looks good now all fan's there. If you tell me Fan1 and Fan2 are in the right position it is ok.
Why i think that the chip is not supported,was because i did't find NCT6795D in the supported list within the readme. So sorry for the confusion.
The sensors output looks now so (many thanks!!!):

```
nct6795-isa-0a20
Adapter: ISA adapter
in0:                    +0.51 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                    +3.42 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.78 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.23 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.60 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.68 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                  1997 RPM  (min =    0 RPM)
fan2:                   696 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                   832 RPM  (min =    0 RPM)
SYSTIN:                 +34.0°C  (high = +127.0°C, hyst =  +0.0°C)  sensor = CPU diode
CPUTIN:                 +29.0°C  (high = +127.0°C, hyst = +105.0°C)  sensor = thermistor
AUXTIN0:                +38.0°C  (high = +127.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +50.0°C    sensor = thermistor
AUXTIN3:                 -2.0°C    sensor = thermistor
SMBUSMASTER 0:          +26.5°C  (high = +127.0°C, hyst = +105.0°C)
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

groeck, 2017-08-05:
Updated README. Thanks for testing!

closed 2017-08-05 by groeck

scorpio810, 2017-08-24:
Nice ! thanks Groeck. ;-)
On B350 Tomakwak :

```
nct6795-isa-0a20
Adapter: ISA adapter
VCore:                  +0.47 V  (min =  +0.00 V, max =  +1.74 V)
VPPM:                   +1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VCC:                   +3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
+12V:                  +12.29 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
VDDCR_SOC:              +0.15 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
DRAM:                   +1.16 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
VBat:                   +3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
Null:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
VIN10:                  +0.00 V  (min =  +0.00 V, max =  +0.00 V)
Chipset 1.05V:          +1.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
+5V:                    +2.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
VDDP:                   +0.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
1.8V+:                  +1.52 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
Chassis Fan 2 Speed:    890 RPM  (min =    0 RPM)
CPU Fan 1 Speed:        588 RPM  (min =    0 RPM)
Chassis Fan 1 Speed:    912 RPM  (min =    0 RPM)
Chassis Fan 4 Speed:   1039 RPM  (min =    0 RPM)
Chassis Fan 3 Speed:    802 RPM  (min =    0 RPM)
PUMP Fan 1 Speed:       993 RPM  (min =    0 RPM)
Motherboard:            +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPU:                    +32.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
Auxillary:              +39.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
VRM:                   -128.0°C    sensor = thermistor
AUXTIN2:                +23.0°C    sensor = thermistor
CPU:                     -1.0°C    sensor = thermistor                                                                                                                                                                                                                        
SMBUSMASTER 0:          +27.5°C                                                                                                                                                                                                                                               
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C                                                                                                                                                                                                                                               
PCH_CHIP_TEMP:           +0.0°C                                                                                                                                                                                                                                               
PCH_CPU_TEMP:            +0.0°C                                                                                                                                                                                                                                               
intrusion0:            ALARM                                                                                                                                                                                                                                                  
intrusion1:            ALARM                                                                                                                                                                                                                                                  
beep_enable:           disabled
```