groeck/nct6775 #54: Fancontrol uses the wrong temperature source with nct6795-isa-0a20

Nicop06, 2017-12-23:
I have an MSI B350 TOMAHAWK. I setup the fans so that they are controlled by the CPU temperature, and it works perfectly in the BIOS.

When I boot my system, the temperature source used to control is the one called "SMBUSMASTER 0", while it should be CPUTIN. I did extensive checks and the CPUTIN temperature is aligned with the one reported by the BIOS and does vary as expected when increasing or decreasing my machine load. As for the "SMBUSMASTER 0", its variation seems completely random lading to high fan speed when the system is idle or low fan speed under load.

I wrote a script to change the temperature source on boot, but I believe it would be easier if the driver used the value reported by the motherboard.

I an using the latest version of the driver from this repository.

groeck, 2017-12-24:
SMBUSMASTER 0 is correct. The chip variant used on MSI boards is an MSI-specific variant which uses that input for the CPU temperature. There is no known means to detect that this is an MSI-specific chip, thus the code doesn't override the name of the sensor.

closed 2017-12-24 by groeck

Nicop06, 2017-12-24:
You are right, the temperature reported by `SMBUSMASTER 0` is correct... unless the fans are controlled by it. Here is the temperature reported by sensor when the fans are using `SMBUSMASTER 0` as a control source (this is what I get after booting the system):

```
fan1:                     0 RPM  (min =    0 RPM)                                             
fan2:                  1681 RPM  (min =    0 RPM)                                             
fan3:                  1006 RPM  (min =    0 RPM)                                             
fan4:                     0 RPM  (min =    0 RPM)                                             
fan5:                     0 RPM  (min =    0 RPM)                                             
fan6:                   900 RPM  (min =    0 RPM)                                             
SYSTIN:                 +36.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode  
CPUTIN:                 +33.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor       
AUXTIN0:                +41.5°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor       
AUXTIN1:               -128.0°C    sensor = thermistor                                        
AUXTIN2:                +18.0°C    sensor = thermistor                                        
AUXTIN3:                 -1.0°C    sensor = thermistor                                        
SMBUSMASTER 0:          +52.0°C
```

And now, if I set the control source to the `CPUTIN` temperature, you can see that the fan speed decreases (I can confirm that from the fan noise) and the `SMBUSMASTER 0` gets aligned with the `CPUTIN` temperature. Only a few seconds elapsed between both output the system is mostly idle except for this github page:

```
fan1:                     0 RPM  (min =    0 RPM)                                             
fan2:                  1459 RPM  (min =    0 RPM)                                             
fan3:                   870 RPM  (min =    0 RPM)                                             
fan4:                     0 RPM  (min =    0 RPM)                                             
fan5:                     0 RPM  (min =    0 RPM)                                             
fan6:                   752 RPM  (min =    0 RPM)                                             
SYSTIN:                 +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode  
CPUTIN:                 +33.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor       
AUXTIN0:                +41.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor       
AUXTIN1:               -128.0°C    sensor = thermistor                                        
AUXTIN2:                +18.0°C    sensor = thermistor                                        
AUXTIN3:                 -1.0°C    sensor = thermistor                                        
SMBUSMASTER 0:          +33.0°C
```

`+33.0°C` is the expected temperature (the one reported by the BIOS). Changing back to `SMBUSMASTER 0` will result in the first output again.

It's as if setting the fan control source to `SMBUSMASTER 0` changes the output of the sensor. In that case I get higher temperature leading to increased noise, but sometimes I get lower temperature which can lead to overheating.

groeck, 2017-12-24:
The temperature displayed as SMBUSMASTER 0 is the raw temperature reported by the CPU. As known with AMD Ryzen 1600X/1700X/1800X CPUs, this is not the real temperature but 20 degrees C higher. The BIOS fan control is aligned to this temperature. If you change it, you'll also have to change the respective temperature curves for automatic fan control, or the system _will_ overheat. On my system I just leave it alone, and it works quite nicely.

Nicop06, 2017-12-25:
Actually, I read somewhere that the latest BIOS update fixed the +20 degrees C (sorry I cannot find the source). This does not explain why the temperature reported by the `SMBUSMASTER 0` sensor changes when I change the control source. By itself, this should be a bug. For reference, here is the script I use to perform the source change:

```
SRC=2
for f in /sys/devices/platform/nct6775.2592/hwmon/hwmon1/pwm?_temp_sel
do
  echo $SRC | sudo tee $f
done
```

Here is what I have in the BIOS when the system has been idle for a few hours:

[image]

And here is what I get right after booting with nothing running:

```
fan2:                  1735 RPM  (min =    0 RPM)
fan3:                  1083 RPM  (min =    0 RPM)
fan6:                   967 RPM  (min =    0 RPM)
SYSTIN:                 +33.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = CPU diode
CPUTIN:                 +35.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
SMBUSMASTER 0:          +59.0°C 
```

The temperature reported by `CPUTIN` after booting is very close to the one reported in the BIOS, while `SMBUSMASTER 0` is off by more than 20 degrees. Now, if I change the temperature source to CPUTIN, you can see that the fan speed is a lot closer to what the BIOS reports:

```
fan2:                  1413 RPM  (min =    0 RPM)
fan3:                   778 RPM  (min =    0 RPM)
fan6:                   688 RPM  (min =    0 RPM)
CPUTIN:                 +33.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
SMBUSMASTER 0:          +33.0°C  
```

Now, if I wait for a few minutes in the BIOS screen, the temperature will rise to 50 degrees (waiting longer will make it rise by only a few degrees). Here is what I get:

[image]

I can reproduce those conditions by compiling the linux kernel with 17 threads for a few minutes and here is what I get:

```
fan2:                  1988 RPM  (min =    0 RPM)
fan3:                   914 RPM  (min =    0 RPM)
fan6:                   810 RPM  (min =    0 RPM)
CPUTIN:                 +50.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
SMBUSMASTER 0:          +80.5°C  
```

Temperature reported by `SMBUSMASTER 0` is 10 degrees higher than what the BIOS reports when taking into account the 20 degrees difference and the fans are 20-30% faster. Now, when setting the source to CPUTIN:

```
fan2:                  1607 RPM  (min =    0 RPM)
fan3:                   915 RPM  (min =    0 RPM)
fan6:                   809 RPM  (min =    0 RPM)
CPUTIN:                 +50.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor                                                                                                                                                                                                                                     
SMBUSMASTER 0:          +50.5°C
```

Now the fan speed is closer to what the BIOS is reporting and so is the `SMBUSMASTER 0`.

Why is `SMBUSMASTER 0` acting so strangely when changing the fans control source? Maybe this is the problem that needs to be fixed.

Anyway, in the meantime using `CPUTIN` as a source gives the expected fan speed while `SMBUSMASTER 0` does not.

Nicop06, 2017-12-25:
Actually, the value reported by `SMBUSMASTER 0` is the one reported by the sensor used to control the pwm1 fan. I wrote a little script to prove that:

```
for i in {1..10}
do
  echo $i | sudo tee pwm1_temp_sel
  sleep 3
  cat temp${i}_input
  cat temp7_input; sensors | grep SMBUSMASTER
done
```

And here is what I get:

```
1                                                                                                                                                                                             
35000                                                                                                                                                                                         
35000                                                                                                                                                                                         
SMBUSMASTER 0:          +35.0°C                                                                                                                                                               
2                                                                                                                                                                                             
33000                                                                                                                                                                                         
33000                                                                                                                                                                                         
SMBUSMASTER 0:          +33.0°C                                                                                                                                                               
3                                                                                                                                                                                             
44500                                                                                                                                                                                         
44500                                                                                                                                                                                         
SMBUSMASTER 0:          +44.5°C                                                                                                                                                               
4                                                                                                                                                                                             
-128000                                                                                                                                                                                       
-128000                                                                                                                                                                                       
SMBUSMASTER 0:         -128.0°C                                                                                                                                                               
5                                                                                                                                                                                             
18000                                                                                                                                                                                         
18000                                                                                                                                                                                         
SMBUSMASTER 0:          +18.0°C                                                                                                                                                               
6                                                                                                                                                                                             
-1000                                                                                                                                                                                         
-1000                                                                                                                                                                                         
SMBUSMASTER 0:           -1.0°C                                                                                                                                                               
7                                                                                                                                                                                             
52500                                                                                                                                                                                         
52500                                                                                                                                                                                         
[... cut ...]
10                                                                                                                                                                                            
0                                                                                                                                                                                             
0                                                                                                                                                                                             
SMBUSMASTER 0:           +0.0°C  
```

groeck, 2017-12-25:
cputin vs. sbusmaster 0 is not a valid reference. Compare with the output of k10temp (from 4.15-rc or from github). Either case, the nct6775 driver does not change the chip configuration on its own; it only reports what the BIOS has configured. Since I don't know what is connected to what on the mainboard, I can not comment on what happens when changing the chip configuration.
If you think there is a problem in the driver, please feel free to examine its source and identify it; for my part all I can say is that I don't have a problem with the same driver on the same motherboard myself. I'll be happy to accept patches.