groeck/nct6775 #55: fan rpm zero Asus QC5000M-ITX/PH board

udovdh, 2018-02-02:
UEFI 'bios' shows 15xx RPM for chasis fan, yet in Fedora 26 Linux:

# sensors-detect 
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASRock QC5000M-ITX/PH
# Kernel: 4.14.16 x86_64
# Processor: AMD A4-5050 APU with Radeon(TM) HD Graphics (22/0/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             Success!
    (driver `fam15h_power')
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT5573D/NCT5577D/NCT6776F Super IO Sensors' Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): YES      
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Using driver `i2c-piix4' for device 0000:00:14.0: AMD Hudson-2 SMBus
Module i2c-piix4 loaded successfully.
Module i2c-dev loaded successfully.

Next adapter: AMDGPU i2c bit bus 0x90 (i2c-0)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x91 (i2c-1)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x92 (i2c-2)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x93 (i2c-3)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x94 (i2c-4)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x95 (i2c-5)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x96 (i2c-6)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: AMDGPU i2c bit bus 0x97 (i2c-7)
Do you want to scan it? (yes/NO/selectively): YES

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT5573D/NCT5577D/NCT6776F Super IO Sensors' (confidence: 9)

Driver `fam15h_power' (built-in):
  * Chip `AMD Family 16h power sensors' (confidence: 9)

Driver `k10temp' (built-in):
  * Chip `AMD Family 16h thermal sensors' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Unloading i2c-dev... OK
Unloading i2c-piix4... OK
Unloading cpuid... OK

[root@vuurmuur linux]# sensors -s
[root@vuurmuur linux]# sensors 
amdgpu-pci-0008
Adapter: PCI adapter
temp1:        +38.0°C  (crit = +120.0°C, hyst = +90.0°C)

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:           N/A  (crit =  13.51 W)

nct6776-isa-0290
Adapter: ISA adapter
Vcore:          +0.92 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +0.18 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:           +3.38 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:          +3.36 V  (min =  +2.98 V, max =  +3.63 V)
in4:            +0.51 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:            +1.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:            +1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:           +3.41 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:           +3.25 V  (min =  +2.70 V, max =  +3.63 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:             0 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
SYSTIN:         +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:         +33.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN:         -17.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
PCH_CHIP_TEMP:   +0.0°C  
PCH_CPU_TEMP:    +0.0°C  
PCH_MCH_TEMP:    +0.0°C  
cpu0_vid:      +0.000 V
intrusion0:    OK
intrusion1:    OK
beep_enable:   disabled

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +38.1°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

groeck, 2018-02-02:
No idea what is happening, and there is not enough context information to proceed. From the "sensors" output it is obvious that there is a sensors.conf entry, but the entry is not provided. It is also unknown if there are any other fans connected to the system, and if moving the fan to a different fan header makes a difference. It may well be that the sensors configuration file somehow disables the fans, but that is just a wild guess.

udovdh, 2018-02-03:
sensors3.conf on Fedora contains:

chip "w83627ehf-*" "w83627dhg-*" "w83667hg-*" "nct6775-*" "nct6776-*" "nct6779-*" "nct6791-*"

    label in0 "Vcore"
    label in2 "AVCC"
    label in3 "+3.3V"
    label in7 "3VSB"
    label in8 "Vbat"

    set in2_min  3.3 * 0.90
    set in2_max  3.3 * 1.10
    set in3_min  3.3 * 0.90
    set in3_max  3.3 * 1.10
    set in7_min  3.3 * 0.90
    set in7_max  3.3 * 1.10
    set in8_min  3.0 * 0.90
    set in8_max  3.3 * 1.10

That is the only spot nct* chips are referenced. I see no fan-related lines here.
There is only one fan connected.
I did not move the fan as I expect both headers to work.  I will try this later today.
The UEFI `BIOS` shows things _do_ work.

udovdh, 2018-02-03:
To be sure, with empty config:

$# sensors -c /dev/null 
amdgpu-pci-0008
Adapter: PCI adapter
temp1:        +29.0°C  (crit = +120.0°C, hyst = +90.0°C)

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:           N/A  (crit =  13.51 W)

nct6776-isa-0290
Adapter: ISA adapter
in0:            +1.00 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +0.18 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:            +3.38 V  (min =  +2.98 V, max =  +3.63 V)
in3:            +3.36 V  (min =  +2.98 V, max =  +3.63 V)
in4:            +0.47 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:            +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:            +1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:            +3.41 V  (min =  +2.98 V, max =  +3.63 V)
in8:            +3.23 V  (min =  +2.70 V, max =  +3.63 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:             0 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
SYSTIN:         +20.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:         +20.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN:         -17.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
PCH_CHIP_TEMP:   +0.0°C  
PCH_CPU_TEMP:    +0.0°C  
PCH_MCH_TEMP:    +0.0°C  
cpu0_vid:      +0.000 V
intrusion0:    OK
intrusion1:    OK
beep_enable:   disabled

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +28.8°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

udovdh, 2018-02-04:
After attaching another fan to the 'CPU_FAN1' CPU fan connector RPMs did appear:
fan2:          1454 RPM  (min =    0 RPM)
This means that we have now two fans connected, one to CPU_FAN1 and one to FAN_FAN1 but only the CPU fan connector does show RPM.

groeck, 2018-02-04:
The configuration information in ASRock's A-tuning utility shows that the system fan is not controlled by the Super-IO controller, but by MMIO. Please feel free to download and install the utility on a Windows system and verify; you'll find the configuration information in the "Conf" subdirectory. There is nothing we can do about this in the nct6775 driver.

closed 2018-02-04 by groeck