groeck/nct6775 #56: NCT6793 - Can't switch PWM mode

DocMAX, 2018-03-26:
On this chip i can't change the mode. It seems it's running PWM mode all the time (pwmX_mode set to 0). Can anybody confirm that DC mode is working on this chip?

My mainboard is ASUS PRIME J3355.

Thanks.

groeck, 2018-03-26:
The primary reason is that the chip, according to its datasheet, only supports DC mode on pwm channel 0. The secondary reason is that the code has a bug and doesn't reject an attempt to set the mode to 0 on other channels, as it should. When that is fixed, the difference will be that "echo 0 > pwm1_mode" will fail with -EINVAL.

DocMAX, 2018-03-26:
What do you mean by "PWM channel"? pwm0_mode? There is no pwm0, just pwm1 and pwm2.

groeck, 2018-03-26:
Ah yes, sorry. pwm counts start with 1. I'll fix the above.

DocMAX, 2018-03-26:
OK, but thats the channel i try to change and it doesn't change to 0 (DC mode).

groeck, 2018-03-26:
"The code has a bug and confuses 0 and 1. It _might_ set the mode to 0 if you write a 1 into the attribute."
Emphasis on _might_ this time.

DocMAX, 2018-03-26:
Indeed, i get a 0 now, but still, i can't control the fan (3pin on chassis fan connector, which is fan1/pwm1).

groeck, 2018-03-27:
The ABI problem should now be fixed. However, that doesn't mean that fan control now actually works in DC mode; it may well be that the board connects the fan control output directly to the fan header's pwm pin. Only the board vendor would be able to answer this. Note that PWM/DC fan control mode can sometimes be configured in the BIOS. I have no idea though if this is the case for ASUS.